solokeys/solo1 #158: Papercuts

nickray, 2019-03-28:
This is the list of currently known issues that are
- not thought to be critical
- not (easily) fixable in a backwards-compatible way

The plan is to fix these if either
- a security critical bug forces us to rollout firmware upgrades for everybody, or
- we move to a new chip (e.g. Cortex-M23 or M33 series), with separate firmware builds

1. U2F counter starts at 2130706432 instead of zero (FIDO2 counter not affected): https://github.com/solokeys/solo/blob/4e21c0bd8ff18c9066b88b549a54289901ae482f/fido2/u2f.c#L250
2. Resident Key display name limited to 32 instead of 64 characters: https://github.com/solokeys/solo/blob/4e21c0bd8ff18c9066b88b549a54289901ae482f/fido2/ctap.h#L104

mutantmonkey, 2019-03-30:
> U2F counter starts at 2130706432 instead of zero (FIDO2 counter not affected): 

This seems like it's more than a minor problem. If I register in a browser that only uses U2F/CTAP1 and then try to authenticate in a browser using CTAP2, won't that cause authentication to fail? It seems like at the very least this same treatment would need to be applied to the CTAP2 counter.

conorpp, 2019-04-01:
Great point!  Will just rip the band aid off now and update U2F counter for good.

onlykey, 2019-05-03:
@conorpp @nickray I am working on porting Solo firmware over to OnlyKey. Its different hardware so unfortunately there are lots of changes that were required (i.e. We use hardware wear leveled EEPROM). A couple of question related to 2. above:

- It looks like the only change needed is to change DISPLAY_NAME_LIMIT to 64 right? 
- Can you recommend any good test resources to test out resident keys? Right now I have OnlyKey supporting 5 RKs and have tested with the FIDO2 conformance tool but didn't find any sites to test it on, this one looks like it supports RK but it doesn't yet - https://fido2.azurewebsites.net/
- If you guys do have to do a firmware update with breaking changes I have some things I would recommend like using uECC_sign_deterministic instead of uECC_sign https://github.com/trustcrypto/libraries/blob/419fe44dbd5ab065f118df31881185f32aab0ee7/fido2/crypto.cpp#L190. I can do a PR for this.

aseigler, 2019-05-03:
With regard to RKs and https://fido2.azurewebsites.net, best I recall is that it does work, but possibly only with certain browsers.

0x0ece, 2019-05-03:
demo.yubico.com also has a demo for RK