solokeys/solo1 #159: Ubuntu 18.10 Udev Rules

charles-d-burton, 2019-03-30:
I just bought a second solokey with version 2.0.0 installed and it's not recognized by the new udev rules  My older key with 1.1.0 still works fine(only in firefox for some reason), but as soon as I try to use the 2.0.0 key nothing happens.  The light never goes gold and the /dev/solokey device is not created.

Here's the dmesg output for the 2.0.0 key:
```bash
[81307.717655] usb 1-1: new full-speed USB device number 14 using xhci_hcd
[81307.867431] usb 1-1: New USB device found, idVendor=0483, idProduct=a2ca
[81307.867437] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[81307.867441] usb 1-1: Product: Solo 2.0.0
[81307.867444] usb 1-1: Manufacturer: SoloKeys
[81307.867447] usb 1-1: SerialNumber: 209C36805548
[81307.870151] hid-generic 0003:0483:A2CA.000C: hiddev0,hidraw1: USB HID v1.11 Device [SoloKeys Solo 2.0.0] on usb-0000:00:14.0-1/input0
```
Here's the output for the 1.1.0 Key
```bash[81692.204154] usb 1-1: new full-speed USB device number 15 using xhci_hcd
[81692.354527] usb 1-1: New USB device found, idVendor=0483, idProduct=a2ca
[81692.354533] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[81692.354537] usb 1-1: Product: Solo
[81692.354541] usb 1-1: Manufacturer: Solo Keys
[81692.354544] usb 1-1: SerialNumber: 0123456789ABCDEF
[81692.359344] hid-generic 0003:0483:A2CA.000D: hiddev0,hidraw1: USB HID v1.11 Device [Solo Keys Solo] on usb-0000:00:14.0-1/input0
```

I'm using the udev rules provided in this repository, installed via the make file.

charles-d-burton, 2019-03-30:
For more info, these are the current udev rules
```bash
# Notify ModemManager this device should be ignored
ACTION!="add|change|move", GOTO="mm_usb_device_blacklist_end"
SUBSYSTEM!="usb", GOTO="mm_usb_device_blacklist_end"
ENV{DEVTYPE}!="usb_device",  GOTO="mm_usb_device_blacklist_end"

ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", ENV{ID_MM_DEVICE_IGNORE}="1"

LABEL="mm_usb_device_blacklist_end"

# Solo bootloader + firmware access
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"

# ST DFU access
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"

# U2F Zero
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="8acf", TAG+="uaccess"
```

conorpp, 2019-03-30:
Can you try installing `solo-python` and see if it is detected?

```
pip3 install solo-python
solo ls
```

If not, can you try with this more simple udev rule?

```
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess", MODE="0660", GROUP="plugdev"
```

Then run:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

xMordax, 2019-04-17:
I have the same problems, Chrome does not work on my Chromeium but it works on my Firefox.
Same goes for my Android LineageOS Chrome.

charles-d-burton, 2019-04-25:
I reflashed the key and everything seems to be working now for me.

closed 2019-04-25 by charles-d-burton