solokeys/solo1 #160: Bricked Solo Hacker

kmad, 2019-03-30:
I bought a Solo Hacker not too long ago and went to try out some of the development tooling using the build guide, and seemed to have corrupted either the bootloader or some other cirtical code.

After updating the udev rules and trying to program the Solo using the supplied Python script I got an odd output (below) and seemed to have bricked my Solo. It is now not recognized at all under Linux or Mac; nothing in dmesg and no LED lights up. I've tried holding the button down during plug-in with the same result. Is there any way to recover from this?

This is the output I received - after this the device was bricked.

`$ python3 tools/solotool.py program targets/stm32l432/solo.hex`

> No Solo device detected.
> erasing...
> downloading 101.63%  08017800 - 08017826 ...
> time: 4761 ms
> verifying...
> reading 203.26%  08017800 - 08017826 ...
> firmware readback verified.

conorpp, 2019-03-30:
Did you put it in bootloader mode or DFU mode?  In bootloader mode, the LED is a fast-flashing yellow.  In DFU mode, the LED is off.  `solo.hex` is just the application with no bootloader, so if you program it while key is in DFU mode, the bootloader will be removed and it will try to boot and brick itself.  I don't think there's a way to restore it without using a STlink or something.

We need to update our docs and remove that script.  The best way to update now is to use our separate solo tool:
https://github.com/solokeys/solo-python

```
solo program bootloader targets/stm32l432/solo.hex
```

conorpp, 2019-04-04:
On second look, `tools/solotool.py` is no longer in the repo and our docs are up-to-date.  Did you use documentation locally on an old checked out version of solo repo?

kmad, 2019-04-08:
I performed this action back when `solotool.py` was a part of the repo (but just got around to posting an issue about it!) 
As it seems it's unlikely I'll be able to recover without STLink I'll just order a new Hacker and close this issue. I'll be the guinea pig on the new update process :+1:

closed 2019-04-08 by kmad

My1, 2019-08-14:
> Did you put it in bootloader mode or DFU mode? In bootloader mode, the LED is a fast-flashing yellow. In DFU mode, the LED is off. `solo.hex` is just the application with no bootloader, so if you [...]

could solo be made in a way that if there's no bootloader we just go into DFU so you can flash it?