solokeys/solo1 #167: Check for broken links in docs updates

ehershey, 2019-04-06:
Maybe as part of CI or docs site updates, I think it would be nice to somehow scan for broken links and give PR feedback or at least prevent publishing them. 

For example in https://github.com/solokeys/solo/pull/163, the relative README URL (../../README.md) looks fine on github but breaks on the live site - https://docs.solokeys.io/solo/ - linking to https://docs.solokeys.io/README.md

nickray, 2019-04-06:
I removed the link to the README; it was a leftover from when there wasn't any actual documentation.

For the link checking in general, that would be great of course.

For context, right now we have a modular documentation with a "container" page https://docs.solokeys.io prepared to have sub-documentation for each of our projects. This does lead to some complication such as the prefixes in https://github.com/solokeys/solo/blob/master/mkdocs.yml#L10-L19, and has broken in the past.

An alternative would be to have all things documentation in a separate repository for a more unified reader experience, at the expense of having docs local to the project they apply to.

nickray, 2019-04-06:
I moved the "stub" or "container" documentation from GitLab to GitHub: https://github.com/solokeys/docs

nickray, 2019-04-06:
There is one horrible papercut: If you visit `https://docs.solokeys.io/solo/` then all is fine. If you visit `https://docs.solokeys.io/solo` then all links are broken. Would be great to fix this this if possible.

ehershey, 2019-04-07:
Yikes! Two quick ideas to fix that are - 1) Use javascript to send browsers a redirect to https://docs.solokeys.io/solo/ if they hit https://docs.solokeys.io/solo without the trailing slash or 2) convert all the links in mkdocs.yml to absolute paths ala /solo/index.md.

Also I found [some notes](https://github.com/mkdocs/mkdocs/issues/738) on link checking mkdocs sites - there's a strict mode in mkdocs that might do it and [awesomebot](https://github.com/dkhamsing/awesome_bot) and [milv](https://github.com/magicmatatjahu/milv) are two third party tools that might be easy to integrate.