solokeys/solo1 #170: Invalid CBOR in GetInfo response

agl, 2019-04-10:
Just poking at this device with Chrome, after flashing firmware 2.1.0:

Chrome rejects the GetInfo response because the CBOR is invalid and thus treats this firmware revision as U2F-only. Specifically, the entries in the CBOR map are incorrectly ordered. See [section 6](https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#ctap2-canonical-cbor-encoding-form) on ordering.

conorpp, 2019-04-10:
Thanks for reporting.  I just pushed a fix for this and tested Chrome is able to register+auth with FIDO2 without falling back to U2F.

Do you have a hacker/reprogrammable Solo and would be able to test on your side?  If so, you can update with my fix.

[solo.hex.zip](https://github.com/solokeys/solo/files/3065116/solo.hex.zip)

```
pip3 install python-solo
unzip solo.hex.zip
solo program bootloader solo.hex
```

0x0ece, 2019-04-11:
Good catch! Just tested this new firmware on Safari 12.1, the future is passwordless :)

[image]

agl, 2019-04-12:
> Do you have a hacker/reprogrammable Solo and would be able to test on your side? If so, you can update with my fix.

I have one on order.

PR #171 (2019-04-12): Fix cbor
Should fix #170.  Most of the CBOR maps encoded by Solo were not ordered in any particular way, so some platforms may fail to parse them since [they are supposed to be ordered](https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#ctap2-canonical-cbor-encoding-form). 

Also adds to python tests to check parsed CBOR maps to be in correct order.  Thanks to @agl for reporting.
closed 2019-04-12 by conorpp