solokeys/solo1 #181: Firmware Update Failed and Stuck - Issues with Chrome and Firefox

royimous, 2019-04-23:
Hello,

There are a few things that have gone wrong. 

tl;dr Solokey wasn't working with Google on Chrome, turns out firmware was out of date. I tried to update with web updater on Firefox and it got stuck. Solokey is now stuck in update mode.

Here's what's happened so far:

* It doesn't work with Google. I tried to add it to my Google account and it wouldn't register the button press. 
* It _did_ work with Github and allowed me to add it to the account. 
* I then realized maybe it needs a firmware update. I found the web update page and it _did not_ work with Chrome. 
* I switched to Firefox and web update seemed to work. It told me I was on version (iirc) 1.1.1. Definitely not the most recent version. So I followed the instructions and it started to update (along with the flashing popups). This failed. It got stuck at 35.29% update progress
* I then installed the local version of web update on my mac and tried to update it. I made a mistake, I think, and held down the button while plugging my Solokey in before I allowed the local version of web update to identify the Solokey. 
* My Solokey is not stuck with the fast blinking yellow light. Even when I unplug it and plug it back in it persists.

How do I get out of this update mode? 

Chrome version: 73.0.3683.103
Firefox version: 66.0.3

Some requests:
* Please update the FAQ with instructions on updating the Firmware and how to get help if something doesn't work.
* Add a guide on what the various blinking lights/colors mean.
* Also add instructions on what to do if the key is stuck in firmware update mode.

Thanks!

royimous, 2019-04-23:
@conorpp damn, impressed you saw this already. Thanks! 👍

conorpp, 2019-04-23:
Thanks for reporting!  What OS are you on?

If it works in Firefox and is  stuck in bootloader mode (flashing yellow LED), can you click on "Advanced mode" and click on one of the flash buttons?  Depends if you have a hacker or Solo device.

Will open separate issues for documentation, I agree, it should be improved soon.

royimous, 2019-04-23:
D'oh, sorry forgot to tell you my OS!

macOS 10.14.4

nickray, 2019-04-23:
Did you tab away during the update? This aborts the update process.

royimous, 2019-04-23:
I did tab away. I think I switch back to Chrome to look over this Github
repo. I didn't know it aborts the process when you do that.

conorpp, 2019-04-25:
Can you try getting it to work on Chrome using "Advanced mode"?  Put key in bootloader mode first and click on flash secure or flash hacker, depending on which device you have.

royimous, 2019-04-25:
It worked! I used Advanced Mode in Chrome (the most recent build 74.0.3729.108) and it re-flashed the solokey and it's now on firmware version 2.2.2. Thanks!

nickray, 2019-04-25:
I added a warning to the effect that tabbing out during updates is VERBOTEN, hopefully this will help others avoid your troubles.

manuel-domke, 2019-04-25:
By the way: Is it expected to get constant "Timeout" popups when using the updater? 
I mean not only during the update process, but on every operation.

Always had, but ignored these, because it otherwise worked fine (for me), but it's a bit annoying.
Using Chrome (different versions) on Linux only.

nickray, 2019-04-25:
This happens if you have very old firmware (e.g. Kickstarter batch). This doesn't have all the necessary extension commands. *Should* go away after updating, let us know if it does not.

royimous, 2019-04-25:
Thanks for all your help @nickray . My Solo is working now 👍

alexjmoore, 2019-07-04:
I am not sure if it is related exactly - I can open a new issue if not.  But I just received some Solo Taps, two things happen for me.

1. When I go to the update site it immediately brings a Chrome pop up saying "Something went wrong" and "The request timed out"

2. If I then close that pop up and click "Inspect Key", it brings up a "Windows Security" prompt to set a PIN, when I try and set one, the key turns red for around 2-3s, but then the same prompt to set a PIN returns, no error at all.  (by the way is there a reference for the LED states?)

I have successfully managed to attach the key to my Google account however, so it does seem functional, and given it is a brand new key I expect it already has fairly recent firmware.

But any thoughts on these two?   is it related? or shall I open a fresh issue.

Running Chrome 75.0.3770.100 64bit in Windows 10 Pro 1903 Build 18362.207

emansih, 2019-07-05:
@alexjmoore same issue. Here's what I did. 

1. Go to `https://update.solokeys.com/`
2. Close the request time out dialog
3. Click on `Inspect Key`
4. Key in a new pin and confirm pin in Windows Security
5. When the key turns red immediately press the button on your key

alexjmoore, 2019-07-08:
Thanks @emansih - that does get me further, but it all seems very glitchy :-)

nickray, 2019-07-08:
The immediate "something went wrong" means you have a firmware that doesn't implement the commands to do version inspection. The in-browser update process is indeed a glorious/horrible hack since WebAuthn does not foresee supporting firmware updates. What's more, browsers are quite different in how the react to this abuse! 😁

For a more stable experience, we suggest using https://github.com/solokeys/solo-python, install and run `solo key update`.

emansih, 2019-07-08:
is it possible to remove the pin after setup?

nickray, 2019-07-08:
You can certainly always run `solo key reset` to remove the PIN, but this also *deletes your keys*. We have an open issue https://github.com/solokeys/solo-python/issues/20.

alexjmoore, 2019-07-30:
Trying again today with the latest firmware, the page loads, it pops up the windows "Making sure its you" dialog for less than a second, then it prompts 'something went wrong, request timed out' from Chrome.

Things are a little better now with 'inspect key', it now cleanly presents the pin dialog and prompts to press the key button, but after that you get the same "something went wrong" dialog.

Firmware: 2.4.2
Chrome: Version 75.0.3770.142 (Official Build) (64-bit)
Windows: 1903 (OS Build 18362.267)

aseigler, 2019-07-30:
> Trying again today with the latest firmware, the page loads, it pops up the windows "Making sure its you" dialog for less than a second, then it prompts 'something went wrong, request timed out' [...]

This happens on Chrome 75 without Solo inserted.