solokeys/solo1 #184: Solokey hacker not turning on after solo program aux enter-dfu

xMordax, 2019-04-24:
I have tried to install the newest bundle-hacker and my Solokey is not turning on anymore.

This is the steps I tried to reproduce:
```
solo program aux enter-bootloader
solo program aux enter-dfu
# powercycle the key (remove and plug in again)
solo program dfu bundle-hacker-2.2.0.hex
solo program aux leave-dfu
# powercycle the key (remove and plug in again)
```
but after the second step solo program aux enter-dfu it does not detect my solo key anymore(it also does not glow anymore)
```
$ solo ls
:: Solos
```

Here is my log when I plug the solokey in:
```
 kernel: [ 5888.125565] pci 0000:39:00.0: [8086:15b5] type 00 class 0x0c0330
 kernel: [ 5888.125653] pci 0000:39:00.0: reg 0x10: [mem 0xd9f00000-0xd9f0ffff]
 kernel: [ 5888.125828] pci 0000:39:00.0: enabling Extended Tags
 kernel: [ 5888.126059] pci 0000:39:00.0: supports D1 D2
 kernel: [ 5888.126078] pci 0000:39:00.0: PME# supported from D0 D1 D2 D3hot D3cold
 kernel: [ 5888.126681] pci_bus 0000:02: Allocating resources
 kernel: [ 5888.126778] pcieport 0000:02:02.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 39] add_size 200000 add_align 100000
 kernel: [ 5888.126818] pcieport 0000:02:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
 kernel: [ 5888.126822] pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
 kernel: [ 5888.126829] pcieport 0000:02:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
 kernel: [ 5888.126832] pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
 kernel: [ 5888.128156] xhci_hcd 0000:39:00.0: xHCI Host Controller
 kernel: [ 5888.128184] xhci_hcd 0000:39:00.0: new USB bus registered, assigned bus number 3
 kernel: [ 5888.129539] xhci_hcd 0000:39:00.0: hcc params 0x200077c1 hci version 0x110 quirks 0x0000000000009810
 kernel: [ 5888.130112] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002
 kernel: [ 5888.130117] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
 kernel: [ 5888.130121] usb usb3: Product: xHCI Host Controller
 kernel: [ 5888.130140] usb usb3: Manufacturer: Linux 4.15.0-48-generic xhci-hcd
 kernel: [ 5888.130143] usb usb3: SerialNumber: 0000:39:00.0
 kernel: [ 5888.130849] hub 3-0:1.0: USB hub found
 kernel: [ 5888.130891] hub 3-0:1.0: 2 ports detected
 kernel: [ 5888.133038] xhci_hcd 0000:39:00.0: xHCI Host Controller
 kernel: [ 5888.133049] xhci_hcd 0000:39:00.0: new USB bus registered, assigned bus number 4
 kernel: [ 5888.133072] xhci_hcd 0000:39:00.0: Host supports USB 3.1 Enhanced SuperSpeed
 kernel: [ 5888.133176] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003
 kernel: [ 5888.133195] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
 kernel: [ 5888.133199] usb usb4: Product: xHCI Host Controller
 kernel: [ 5888.133203] usb usb4: Manufacturer: Linux 4.15.0-48-generic xhci-hcd
 kernel: [ 5888.133206] usb usb4: SerialNumber: 0000:39:00.0
 kernel: [ 5888.133772] hub 4-0:1.0: USB hub found
 kernel: [ 5888.133808] hub 4-0:1.0: 2 ports detected
 upowerd[1674]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb3/3-0:1.0
 upowerd[1674]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb3
 upowerd[1674]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb4/4-0:1.0
 upowerd[1674]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb4
 kernel: [ 5888.472506] usb 3-1: new full-speed USB device number 2 using xhci_hcd
 kernel: [ 5888.622538] usb 3-1: New USB device found, idVendor=0483, idProduct=df11
 kernel: [ 5888.622562] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 kernel: [ 5888.622567] usb 3-1: Product: STM32  BOOTLOADER
 kernel: [ 5888.622572] usb 3-1: Manufacturer: STMicroelectronics
 kernel: [ 5888.622576] usb 3-1: SerialNumber: 207D36905548
 mtp-probe: checking bus 3, device 2: "/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb3/3-1"
 mtp-probe: bus: 3, device: 2 was not an MTP device
 fwupd[6453]: DFU version 0x0000 invalid, v1.1 assumed
 upowerd[1674]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:02.0/0000:39:00.0/usb3/3-1
```

I am not sure how to proceed, can I do somehow a factory reset?

xMordax, 2019-04-24:
If I try to leave-dfu it throws an ```usb.core.USBError: [Errno 13] Access denied (insufficient permissions)```

conorpp, 2019-04-24:
Those steps are correct.  After entering DFU, it will not appear as a Solo and the LED will be off.  It will be in this mode until `solo program aux leave-dfu` is run.  It's important that the firmware programmed is correct before running `solo program aux leave-dfu` or the device might get bricked.  It looks like the device is still in DFU mode, from the log you've posted.

xMordax, 2019-04-24:
Hello @conorpp ,

when I try to run ```solo program dfu bundle-hacker-2.2.0.hex``` or ```solo program aux leave-dfu``` I get the following error:
```
$ solo program dfu bundle-hacker-2.2.0.hex
Traceback (most recent call last):
  File "/home/agantar/.local/bin/solo", line 11, in <module>
    sys.exit(solo_cli())
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/agantar/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/agantar/.local/lib/python3.6/site-packages/solo/cli/program.py", line 64, in dfu
    dfu = solo.dfu.find(serial, attempts=connect_attempts)
  File "/home/agantar/.local/lib/python3.6/site-packages/solo/dfu.py", line 30, in find
    dfu.find(ser=dfu_serial, dev=raw_device, altsetting=altsetting)
  File "/home/agantar/.local/lib/python3.6/site-packages/solo/dfu.py", line 108, in find
    self.dev.set_configuration()
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/core.py", line 869, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/core.py", line 147, in managed_set_configuration
    self.managed_open()
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "/home/agantar/.local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)
```

xMordax, 2019-04-24:
I realized that I have permission problems with the mounting of the solokey, I executed it as root and it worked. 
This is the udev rule I have been using:
```
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess", MODE="0660", GROUP="plugdev"
```

MaPePeR, 2019-04-24:
I'd guess you need to be a member of the group `plugdev` to access the device with that rule?

conorpp, 2019-04-24:
Check our reference udev rules here:

https://github.com/solokeys/solo/blob/master/udev/70-solokeys-access.ruleshttps://github.com/solokeys/solo/blob/master/udev/70-solokeys-access.rules

The part for DFU specifically is 

```
# ST DFU access
SUBSYSTEM=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", TAG+="uaccess"
```

DFU doesn't use a HID interface and its product ID is different.

xMordax, 2019-04-24:
Will test it out with the next update

nickray, 2019-04-25:
> I'd guess you need to be a member of the group `plugdev` to access the device with that rule?

Current versions of udev support the `uaccess` tag. The group `plugdev` access method is only for out-of-date udev versions, hence we have two (minimal) udev rule sets.

My1, 2019-09-09:
> It's important that the firmware programmed is correct before running `solo program aux leave-dfu` or the device might get bricked

would it be possible to add something that automatically kicks solo back into DFU if it cannot properly run after DFUing, for example that after DFU a counter is created and it tries to start a few times and if that doesn't work, back to DFU, to prevent bricking?