solokeys/solo1 #187: Debricking instructions from misflashes

chrysn, 2019-04-30:
I've probably put one of my previously unprogrammed solokeys into a "bricked" mode from which I can't recover with USB alone (swapping out the "bootloader" with "dfu" from "solo program bootloader targets/stm32l432/solo.hex" just 'cause the device is in DFU mode and not in bootloader mode probably wasn't the brightest of ideas in hindsight).

I haven't found any explicit debricking instructions and would assume that there are two viable approaches:
* Shorten some pins when inserting the device to make it boot into DFU again. If that option exists, which vias or test points are there to shorten?
* Wire up the labelled GND, 3V1, SWCLK1 and SWDIO1 to an OpenOCD debugger. Is there a target description and recovery flash file somewhere that can be used in such situations?

nickray, 2019-04-30:
Oh noes :/ There is currently no easy answer.

> * Shorten some pins when inserting the device to make it boot into DFU again. If that option exists, which vias or test points are there to shorten?

Unfortunately, that option does not exist. Agree that in a future revision it would be a good idea to add a solder bridge to the `BOOT0` pin (setting `nSWBOOT0 = 1`) so you can easily get to DFU in such a situation. (Our "secure" build enables ROP level 2, which disables DFU irreversibly anyway, so no safety risk).

> * Wire up the labelled GND, 3V1, SWCLK1 and SWDIO1 to an OpenOCD debugger. Is there a target description and recovery flash file somewhere that can be used in such situations?

This works, e.g. with an ST-LINK/V2, the target is is [stm32l4x](https://github.com/solokeys/solo-bsc/blob/master/openocd.cfg). But you'll either have to solder wires to the labeled pads, or [create a custom connector](https://conorpp.com/blog/3d-printing-a-programming-jig-and-embedding-pogo-pins).

I think/hope you have two keys? If you build a firmware to [start at 0x8005000](https://github.com/solokeys/solo/blob/master/targets/stm32l432/linker/stm32l4xx.ld) (with the intention to jump via our bootloader), beware that you get jumped to in voltage "range 2" ([RM0394](https://github.com/solokeys/solo-bsc/blob/master/ref/reference-manual-RM0394.pdf), p.133), the low-power range. In particular, the USB peripheral can't run (and maximal clock speed is lower, and flash latencies are different). This is because the same bootloader has to work for the Solo Tap, powered via NFC field.

Development currently is much easier using the [NUCLEO-L432KC](https://www.st.com/en/evaluation-tools/nucleo-l432kc.html), or even on PC using the [UDP bridge](https://github.com/solokeys/solo/#developing-solo-no-hardware-needed).

conorpp, 2019-05-01:
Yeah probably best not to use DFU for development.  Is it still in DFU mode?  You can fix it by downloading [bundle-hacker-2.2.2.hex](https://github.com/solokeys/solo/releases/download/2.2.2/bundle-hacker-2.2.2.hex) and running `solo program dfu bundle-hacker-2.2.2.hex`.  Then run `solo program aux leave-dfu` to boot.

Otherwise, if you have the inspiration, I'd just solder wires to the pads on the bottom and connect to an ST Link.

Regarding voltage and flash latencies, programming edited versions of the Solo firmware should be fine since initialization handles increasing the clock rate properly from those states.