solokeys/solo1 #189: The master secret is not persisted after CTAP_RESET

rgerganov, 2019-05-03:
At the end of `ctap_reset()` we call `crypto_reset_master_secret()` which resets the master secret to some random value. The new master secret is not persisted, so all credentials generated after this point will be lost when the key is plugged out, no?

conorpp, 2019-05-25:
That's correct.  It's meant to follow the reset behavior specified [here](https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#authenticatorReset).

conorpp, 2019-05-26:
The `crypto_reset_master_secret()` I think is irrelevant.  The key bits are generated in `ctap_state_init()` and persisted.  They are loaded  into the crypto side with `crypto_load_master_secret(STATE.key_space);`.

rgerganov, 2019-05-27:
No, it is not irrelevant. Consider the following steps to reproduce:
1. Plug the solo key
2. Run `solo key reset`
3. Register the key on a website
4. Unplug the key and then plug it again
5. Trying to authenticate on the same website fails

Performing the above steps with `hidg` emulation gives `[ERR] fido2/ctap.c:1155: Error, no authentic credential`. This is because the credential generated on step 3 is not generated with the persisted master secret but a transient one. Hope this makes it clear.

conorpp, 2019-05-27:
I see what you mean, thanks for outlining.  I think the best fix would be to replace `crypto_reset_master_secret` with `crypto_load_master_secret`, so it's loaded with the new persisted key.

closed 2019-05-28 by conorpp