solokeys/solo1 #198: Problem updating Solo: CTAP error: 0x01 - INVALID_COMMAND

woakas, 2019-05-23:
I have a problem for update my Solo,  Any idea about this error?

```
solo key update --secure 
We are about to update with the latest Solo Secure firmware.
Please confirm that the connected Solo key is a Solo Secure [y/N]: y
Wrote temporary copy of firmware-secure-2.3.0.json to /tmp/tmp8nkvoy6h.json
sha256sums coincide: 9ef2a2af37ab079311564ba956671d79dc6bf887d43380ea45213e2a83e91926
erasing firmware...
problem flashing firmware!
CTAP error: 0x01 - INVALID_COMMAND
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/solo/cli/update.py", line 212, in update
    sig = solo_client.program_file(firmware_file)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 323, in program_file
    self.write_flash(i, data)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 189, in write_flash
    self.exchange(SoloBootloader.write, addr, data)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 138, in exchange_hid
    data = self.send_data_hid(SoloBootloader.HIDCommandBoot, req)
  File "/usr/local/lib/python3.7/dist-packages/solo/client.py", line 133, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "/usr/local/lib/python3.7/dist-packages/fido2/hid.py", line 101, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```

nickray, 2019-05-24:
Are you in bootloader mode? Meaning, is your LED flashing? If not, press button while inserting, keep pressed until it flashes, and try again please.

woakas, 2019-05-24:
Ohhhh perfect, thanks so much. it works like a charm

closed 2019-05-24 by nickray

youiopmop, 2019-05-28:
had this issue with fido2 0.6.0 on python3.7, downgraded to 0.5.0 with pip3 and solo worked like a charm

nickray, 2019-05-28:
@youiopmop yours is a separate issue, which has been resolved: https://github.com/solokeys/solo-python/pull/25

Feel free to update with `pip -U solo-python`, which will pull in solo-python 0.0.11 and fido2 0.6.0, which in combination should work again.