solokeys/solo1 #203: Different behaviour for Facebook login on Firefox and Chrome

armsp, 2019-06-04:
I received my SoloKey today and for starters I decided to enable 2FA on my Facebook account (using Firefox). I was able to do that successfully. However I am **not** able to login to my account via Firefox because Firefox shows the following screen after I enter my username and password -
[image]
Pressing the button on the key doesn't do anything as expected.  

On the other hand with Chrome I get the following and it works -
[image]
My preferred browser is actually Firefox. Any idea how do I make Firefox work the same way as Chrome ?
My OS is Ubuntu.

youiopmop, 2019-06-06:
In Firefox U2F is disabled by default, to enable it you need to go to `about:config` in Firefox and change `security.webauth.u2f` from false to true.

Some websites like GitHub say you need to use Chrome for U2F, but it works fine by changing your user-agent just for the site, you can do that by using something like `general.useragent.override.github.com` in about:config and putting a chrome user-agent there.

armsp, 2019-06-06:
@youiopmop  as I had already mentioned, I had already set the fields to `true`, as a matter of fact, they are set to `true` by default on latest versions of Firefox. 
> I have already setup the udev rules as well as the corresponding fields are set to True in Firefox.  

Following is my current setup that still doesn't work the way its supposed to -
[image]

JackGruber, 2019-06-07:
@armsp 

* Add `general.useragent.override.facebook.com` as new setting
* Use `Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.121 Safari/537.36` as value

for Github this works fine.

armsp, 2019-06-07:
@JackGruber thanks a lot. It works as expected after the changes.

closed 2019-06-07 by armsp