solokeys/solo1 #204: Windows - how to develop without hardware?

KrishnaPG, 2019-06-08:
The ReadMe has a [section](https://github.com/solokeys/solo#developing-solo-no-hardware-needed) describing a case of developing for this without the hardware. Unfortunately it is not clear how to achieve it. Any information in this regard would be of great help.

For example, 
- is there any simulator or some kind of application on windows / Linux that we can run (which simulates the hardware device), so that we can build and test applications that are centered around the keys' functionality?

The instructions in the aforesaid section says, run `make all`. Unfortunately windows / Visual studio does not come with `make`. Not sure how to build this code with Visual Studio. The comment [here](https://github.com/solokeys/solo/issues/113#issuecomment-467723107) indicates that others have already used VS to build this. Appreciate any pointers on how it was done.

In this regard, one suggestion / request is, to adapt `CMake` as the cross-platform build, if possible.

- If one needs to build a web/mobile app that uses these keys as the `keystore manager`, how to build/test such app without really having a hardware key at hand (during the development phase)?

nickray, 2019-06-08:
I think you can install make with chocolatey or scoop.

KrishnaPG, 2019-06-09:
Thank you @nickray 

Assuming that we use the `chocolatey` or similar and get the code built, would we then be able to:

- use the built executable as the `keystore`, similar to the USB device, and avail all the functionalities of the hardware device for testing (without really having a hardware device) ?

The goal is to be able to create / store few keys and check how the whole webAuth flow works, and build /test apps that can use such flow.

0x0ece, 2019-06-09:
@KrishnaPG not really.

The `pc` implementation of Solo listens to a UDP port for "usb" packages. The goal is to let you *develop our firmware* without the hardware, and when you run the tests under the hood we're using a custom python-fido2 library that sends messages over UDP instead of USB.

If you're writing python programs (e.g. a FIDO2 server), then yes, you can reuse this library and interact with the executable.

However if you want to just make a webapp, you'd need to change your browser or your OS, to create an extension or driver that fakes a usb interface and instead sends data via udp. This is currently not possible.

KrishnaPG, 2019-06-10:
Thank you @0x0ece That makes things clear.

Then I am wondering, what is the procedure for creating apps that can take advantage of the Solo keys (without really having a key or two at hand)?

For example, I would really love to provide my customers the ability to use the Solo keys with our app. It is expected that the customers would have the Solo keys, but the developers (of the web/mobile app) do not have the keys at hand.
- In such scenario how can we get ahead with the development (without really having to get into the logistics of hardware procurement, the time delays and cost etc. worries)?

Is there some driver or simulator that can be used for this purpose? It would really really help the developers build the ecosystem around these keys.

alphathegeek, 2019-07-08:
I find a raspberry pi zero works great. It's all of $5.00 and runs the solo application. There is already support for this in the solo codebase, since it can use /dev/hidg.

Failing that, it's up to the app developer to use usb over udp to emulate the key. You need a server running, preferably on a linux machine. the usb specific calls are already replaced with "udp_send" and "udp_recv".

stappersg, 2019-10-17:
Here a happy Linux user.

I have been told that recent versions of Microsoft Windows have something that is called `Windows Subsytem for Linux`, or just `wsl`.

Visiting Microsoft.com got me redirected to a local branch of MSFT and did a search on `Linux`
  https://www.microsoft.com/nl-nl/search?q=Linux

Got several results, one being https://www.microsoft.com/nl-nl/p/debian/9msvkqc78pk6 it says
> With this app you get Debian for the Windows Subsystem for Linux (WSL). You will be able to use a complete Debian command line environment containing a full current stable release environment. Please set up WSL first by following the guide at https://docs.microsoft.com/en-us/windows/wsl/install-win10
>After the installation you can start the WSL console by either clicking the "Debian" tile in the start menu or by typing "debian" in powershell or cmd.
>The source code for this application can be found at the Debian repository: https://salsa.debian.org/rhaist-guest/WSL

I would like to know if `wsl` is a solution that this issue is about. Such as if `make` is available.
Thing I do know is that default installs have no C-compiler installed. Neither is `git` installed.
I think this is needed
```
make --version  # either version information or a "not found"
sudo apt update  # fetches package meta data
sudo apt install git build-essential  # fetches and installs programs
```
to get you started.

Please report your milage

```