solokeys/solo1 #206: Solokey hacker not turning on after solo program aux leave-dfu (Warning: OSError with write_page)

llange, 2019-06-21:
I updated my Solokey hacker from FW 2.1.0 to FW 2.3.0 using `solo key update --hacker` and it worked perfectly.
Then I wanted to upgrade the bootloader from 2.1.0 to 2.3.0.
For this, I have tried to install the newest bundle-hacker (2.3.0) and now my Solokey is not turning on anymore.
(Looks a lot like #184 but failure not exactly at the same point)

Here is a log of the steps I took since I received the key:

# Successful FW upgrade
```
$ solo key version
2.1.0
$ solo key update --hacker
Not using FIDO2 interface.
We are about to update with the latest Solo Hacker firmware.
Please confirm that the connected Solo key is a Solo Hacker [y/N]: y
Wrote temporary copy of firmware-hacker-2.3.0.hex to /var/folders/1k/d00j9f2s0xjb54z0gjxj3nxm0000gr/T/tmpnga4tft9.hex
sha256sums coincide: c9cca58c3e9a32705494f5a15ed7e9a5212f0acce26845362724ff5972f82431
erasing firmware...
updated firmware 100%
time: 6.83 s
bootloader is verifying signature...
...pass!

Congratulations, your Solo Hacker was updated to the latest firmware version: 2.3.0
$ solo key version
2.3.0
```
# Checking bootloader version
```
$ solo program aux bootloader-version
Not using FIDO2 interface.
2.1.0
```

# Unsuccessful bootloader upgrade
(I followed instructions of #113, in "Bundle Update for Solo Hacker")

```
$ solo program aux enter-bootloader
Not using FIDO2 interface.
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
$ solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)
# I did powercycle the key (remove and plug in again)
$ solo program dfu bundle-hacker-2.3.0.hex
erasing...
$ solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)
# I did powercycle the key (remove and plug in again), and from there no LED, no sign of life, no detection of the key
$ solo ls
:: Solos
$ solo program dfu bundle-hacker-2.3.0.hex
Traceback (most recent call last):
...
Exception: no DFU found
$ lsusb
Bus 020 Device 036: ID 05ac:8600 Apple Inc. iBridge
Bus 000 Device 001: ID 1d6b:ISPT Linux Foundation USB 3.0 Bus
Bus 000 Device 001: ID 1d6b:CIAR Linux Foundation USB 3.1 Bus
Bus 001 Device 001: ID 1d6b:CIAR Linux Foundation USB 3.1 Bus
$ solo program aux bootloader-version
Traceback (most recent call last):
....
solo.exceptions.NoSoloFoundError: no Solo found
```

So it looks that at step `solo program aux leave-dfu`, an error occurred (is the the cause, or is it just the consequence of a previous defect during the flash) ?

Could you please check if the output gives any indication of what has happened ?

Context:
- MacBookPro13,3
- macos Sierra 10.12.6 (16G2016)
- virtualenv Python 3.7.3
- solo 0.0.11
- it was a Solo Hacker USB-A, PCB rev 1.2

I guess I need to build or find a [programming jig](https://conorpp.com/blog/3d-printing-a-programming-jig-and-embedding-pogo-pins) now...

llange, 2019-08-22:
Following-up:
(short story) I managed to revive the SoloKey

Long story:
- I could not make a programming jig so I soldered extension wires to the GND, SWDIO, SWCLK test pins to a ST-LINK V2 clone.
- I used the stlink CLI (https://github.com/texane/stlink) to identify, read flash and program flash (Note: First plugged the solokey to a +5V USB ; then connected the ST-LINK V2 to the computer):
```
$ st-info  --probe
Found 1 stlink programmers
 serial: ************************
openocd: "\x**\x**\x**\x**\x**\x**\x**\x**\x**\x**\x**\x**"
  flash: 262144 (pagesize: 2048)
   sram: 49152
 chipid: 0x0435
  descr: L43x/L44x device

$ st-flash read ./flash 0 0x40000

$ st-flash reset
st-flash 1.5.1
2019-08-22T13:54:52 INFO common.c: Loading device parameters....
2019-08-22T13:54:52 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2019-08-22T13:54:52 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes

$ st-flash erase
st-flash 1.5.1
2019-08-22T13:54:57 INFO common.c: Loading device parameters....
2019-08-22T13:54:57 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2019-08-22T13:54:57 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
Mass erasing

$ st-flash --reset --format ihex write bundle-hacker-2.4.3.hex
st-flash 1.5.1
2019-08-22T13:55:19 INFO common.c: Loading device parameters....
2019-08-22T13:55:19 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2019-08-22T13:55:19 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes
2019-08-22T13:55:19 INFO common.c: Attempting to write 231456 (0x38820) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x08038800 erasedEraseFlash - Page:0x131 Size:0x800
2019-08-22T13:55:23 INFO common.c: Finished erasing 114 pages of 2048 (0x800) bytes
2019-08-22T13:55:23 INFO common.c: Starting Flash write for F2/F4/L4
2019-08-22T13:55:23 INFO flash_loader.c: Successfully loaded flash loader in sram
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 2080
2019-08-22T13:55:28 INFO common.c: Starting verification of write complete
2019-08-22T13:55:29 ERROR common.c: Verification of flash failed at offset: 30720
stlink_fwrite_flash() == -1

$ st-flash read ./newflash 0 0x40000
st-flash 1.5.1
2019-08-22T13:55:39 INFO common.c: Loading device parameters....
2019-08-22T13:55:39 INFO common.c: Device connected is: L43x/L44x device, id 0x10016435
2019-08-22T13:55:39 INFO common.c: SRAM size: 0xc000 bytes (48 KiB), Flash: 0x40000 bytes (256 KiB) in pages of 2048 bytes

```

In spite of the error during verification, the key was now revived (pulsing led, visible to `solo ls`, etc...).
To be extra sure, I reflashed it fully (bootloader + application) from within DFU and it went ok without any verification error.

What's surprising, is that when I read the flash it was completly empty - which of course explains the 'bricking'.
It seems like the initial programming process stopped after "erasing" (which is visible in the console output in the beginning of this issue and never went to programming).

Here is the output of a (working) programming process:
```
$ solo program aux enter-bootloader
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.

$ solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)

# I did powercycle the device

$ solo ls --all
:: Solos
:: Potential Solos in DFU mode
***************

$ solo program dfu bundle-hacker-2.4.3.hex
erasing...
downloading 100.00%  08038800 - 08038828 ...
time: 5323 ms
verifying...
reading 100.00%  08038800 - 08038828 ...
firmware readback verified.

$ solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)
Segmentation fault: 11
```

Despite the final "Segmentation fault: 11" everything is ok.

Notes:
* I tried to program with openocd with no luck

wwwrashiade, 2019-10-13:
> * I could not make a programming jig so I soldered extension wires to the GND, SWDIO, SWCLK test pins to a ST-LINK V2 clone.

I'm actually trying to reprogram a Solo Hacker where I flashed a secure build onto.
I have a STLINK/V3SET, so I can't use the CLI you mentioned. Did you just wire up GND, SWDIO and SWCLK? I can't connect with STM32CubeProgrammer using SWD. Did it really work without a reset-line or did you put the solokey into reset with something else?
Thanks for your help.

nickray, 2019-10-13:
If you really flashed a secure build, you're out of luck - it turns on ROP level 2 (irreversibly) and you can't debug anymore.

wwwrashiade, 2019-10-13:
Oh, and I thought that "irreversibly" just meant "irreversibly as a normal user without a programmer" and you get rid of it with a full chip erase or something like Atmels high voltage programming but this seems to be really irreversible without a new chip :/
Thanks for your response

m0ellemeister, 2020-01-21:
> If you really flashed a secure build, you're out of luck - it turns on ROP level 2 (irreversibly) and you can't debug anymore.

@nickray could you please put this information into the documentation for "switching from hacker to secure firmware". Thx.

asnelt, 2020-03-01:
Just to confirm that using the same procedure as described by @llange also works for reviving a Somu Hacker. The Somu cannot be plugged in while having the wires soldered, so I soldered the 3.3 V wire as well for powering the Somu. For the flash writing operation, I had to power cycle before the operation or it would fail. I'm not sure whether this is the case for every Somu or whether this was due to my setup.