solokeys/solo1 #209: Nfc  test

Zjemm, 2019-06-26:
Hi,

I finally received a solo USB A and a solo tab USB C

They both work when plugged in and pressing the key on my laptop. But now I want to test nfc on my Android phone but I can't seem to get it working.

Nfc works on the phone because I pay with nfc at shops.

How can I test nfc on my Android with for example GitHub or Gmail.

I also installed a nfc app that can read nfc chips. So when I scan my bank card it read the chip. But with to solo, nothing happens.

Any advise?

gillesw, 2019-06-27:
I have the same trouble : NFC doesn't work on solo tap USB-C
OnePlus 6T A6013_41-190528 build (latest)

Zjemm, 2019-06-27:
For me it's the OnePlus 3T Android 9.0.3

martinpaljak, 2019-06-27:
I get 6f00 (generic exception) with tap version on the u2f coffe test site with android. It also is “looping” so seems to sort of crash the connection repeatedly.

FabianHenneke, 2019-06-27:
Not working on the OnePlus 5, although it works with YubiKeys and my Solo Tap passes the FIDO test suite with an external NFC reader.

Zjemm, 2019-06-27:
Alright so at least im not the only one. So this must be something. Or maybe a combination with OnePlus

mrvantage, 2019-06-28:
Similar problems here on my OnePlus 3. I am able to read the solo tap using the NXP taginfo app, but only without the sleeve on.

I've tried using the keys with https://u2f.bin.coffee, but I was unable to register it via NFC

martinpaljak, 2019-06-28:
@mrvantage what was the error on the test website?

Tumeez, 2019-06-29:
I also thought that my Solo Tap's are broken, because no react with Oneplus 3. They already know that problem, because I contact them to directly.

tl:dr Solo Tap NFC doesn't work with my Oneplus 3

Zjemm, 2019-06-30:
Well now there is also a GitHub issue, so more people can see

Mincka, 2019-07-01:
Same thing here, I contacted Solo on Twitter but no more answers...

Got response: `{ "errorCode": 1, "errorMessage": "Low level error 0x6f00" }` on https://u2f.bin.coffee/ 
Not working on https://webauthn.io/ neither (with or without rubber protection).
Looks ok on NXP TagInfo.

Tested on Samsung Galaxy A8 (SM-A530F) with Android 9 / One UI 1.0 up-to-date.
Same issue with multiple keys (Type-A or Type-C).

Also tested on Samsung A5 2016 after NFC read, same error.
No NFC field activation at all on Honor 8 and Samsung Galaxy S9.
Powered by a USB-C port from a laptop: same error in the JavaScript response on the mobile.

Tried multiple times, with multiple keys, everywhere on the back. 🙁

They work as expected when plugged to a USB port, but the NFC is not.

manuel-domke, 2019-07-01:
So until now there were only reports regarding "OnePlus" branded smartphones. 

If this affects more manufacturers (or Solo Tap's), I guess there is a bigger issue,Hopefully they can fix it it the firmware. 

Are there any smartphones where it is confirmed working? I have a Tap, but no other NFC-capable device (yet). But maybe I can get hands on some colleague's phones...

Mincka, 2019-07-01:
A colleague told me that he paired his Solo Tap USB-A, plugged on his laptop, to his Google account (nothing new at this stage).

Then, he told me that he managed to authenticate on Google, on his phone, using NFC (on the second try).

Still, he has the same error message (`Low level error 0x6f00`) on https://u2f.bin.coffee/.

Mincka, 2019-07-01:
Oh, some progress  here... I was unable to test the same scenario since I cannot pair my Solo Tap USB-C on my laptop because I have no USB-C or adapter.

However, it made me think that I did not try to combine a U2F registering through USB and an authentication through NFC on https://u2f.bin.coffee/... And it worked, on the third try... 🎉

I tried again and it never works on the first time, but still, the sign in is working at some point. However, the registering over NFC never work. Hope it helps...

manuel-domke, 2019-07-01:
But only on the Galaxy A8? 
The A5, G9 and "Honor 8" (whatever this is) don't react at all, like described for the OnePlus's ?

Zjemm, 2019-07-01:
Don't think it's a OnePlus issue then

dschuermann, 2019-07-01:
I received my Solo Tap (+USB-C) and it works in Chrome with https://u2f.hwsecurity.dev and https://webauthn.hwsecurity.dev .

We implemented workarounds for https://github.com/solokeys/solo/issues/214 and https://github.com/solokeys/solo/issues/213 and now it also works with our native SDK from https://hwsecurity.dev

Mincka, 2019-07-01:
Exactly same behaviour on u2f.hwsecurity.dev and webauthn.hwsecurity.dev: registration only possible with USB (however error code 1 or NotReadableError) and authentication works about 1 time over 3...

I could accept (tolerate) the need to plug for registration but the authentication rate failure is too high.

At least, the NFC field is activated for my phone, on others, there is no detection at all (these phones work perfectly fine for other NFC devices or badges).

conorpp, 2019-07-01:
I've just done some testing on **Pixel, Nexus, Samsung S9, and LG phones**.  I haven't tried OnePlus.

It seems that a lot of **LG** phones simply don't support [ISO 14443A, which is the NFC protocol](https://en.wikipedia.org/wiki/Near-field_communication#Standards) used by Tap.  The LG phone I used wasn't able to read the NDEF data even while Tap was USB powered.  I suspect some OnePlus models have a similar problem.

For **Samsung S9**, it took me a while to get it to work with a U2F test site via NFC, around 60-120s before a registration succeeded.  I couldn't figure out a consistent method for improving that.  I think the RF coupling is too poor.  I suspect this issue extends to most Samsung devices.  NFC-Registering while USB powered I found a firmware bug, but fixing that, it works fine; but no improvement in passive operation.  Will push bug fix soon.

**Pixel** and **Nexus** 6 devices worked fine.  I think recent Motorola and Google devices give the best performance.

I believe most issues stem are stemming from poor RF coupling with some phone models.  Using only authentication instead of registering *should* help since it takes less time and power.  If I can find any potential firmware improvements, I'll work on them.  In any event, we/@solokeys/team-solokeys are planning a new HW revision, and one of the improvements will be significantly reducing power and timing for passive operation.

Tumeez, 2019-07-01:
Will new HW rev be free to us who have a problem at the moment?

0x0ece, 2019-07-02:
Personally, I think we have opportunities to fix via firmware upgrade -- if not we'll certainly explore a hw rev. But I'm positive we can optimize the boot, or play with advanced uses of the AMS chip. It'll just require some time to test and explore options.

It's good to collect feedback about non-working devices.

Zjemm, 2019-07-02:
> I've just done some testing on **Pixel, Nexus, Samsung S9, and LG phones**. I haven't tried OnePlus.
>
> It seems that a lot of **LG** phones simply don't support [ISO 14443A, which is the NFC [...]

i have registered the solo USB-C Tab via usb-c (laptop) on my github account, but still then authentication on the android via NFC (oneplus3t  chrome) does not work.

when plugged in to a laptop for power and then trying NFC, it does read the key but auth fails.

seems indeed like a power issue. I hope you guys can fix it via firmware, otherwise NFC is completely useless in my situation

another thing i noticed that when plugging in the tab on my usb-c on my laptop, the led lights up. but when plugging in on my phone it does not. 
I plugged it into an LG phone and there it did light up
and the USB settings popup from android showed as: Power supply (translated from dutch)

my oneplus3t asks me what to to when plugging in a usb-c cable connected to my laptop (charging only, or file transfer e.t.c.) but it does not ask me when plugging in the solokey in the phone.
also there is no "power supply" option. so the oneplus does not power the key

so it **also** seems that the tab via usb-c does not work on the oneplus

dschuermann, 2019-07-03:
So, some information how Android does FIDO: Chrome is using Google's proprietary FIDO implementation that is shipped in their Google Play Services. I think, we can do a lot better regarding UX and NFC/USB backend. That's why we implemented our own native FIDO stack on Android. Also, this allows us to support special use cases based on customer requirements (such as PDF signing).

Long story short, I just pushed an update to our FIDO Sample (now version 1.6) on Google Play (could take up to 4 hours until you get it): https://play.google.com/store/apps/details?id=de.cotech.hw.fido.example

You can try our implementation with your phone and report if it works better than Chrome. I would be interested in your results.

Also, for some devices, we show where the best NFC spot is located on the back of your device, this may help:
[image]

Zjemm, 2019-07-03:
Nice but first we need an nfc fix, because right now nfc does not work on several Android devices

Also can more people confirm when plugged the usbc in a OnePlus, that it does not receive power?

Mincka, 2019-07-03:
@dschuermann:
After about 100 NFC registering attempts on Galaxy S9: 3 times: "INS_NOT_SUPPORTED", 96 times: vibration but nothing, 1 time: successful registration (really unexpected, first time happening with NFC).
Same results when powered from another phone with USB-C.

However, the authentication works better with your app than Chrome, I'd say authentication succeed about 8/10 tries.

Side note regarding the app: the key has to be plugged BEFORE taping on buttons (register / authenticate). It does not work if the help message is displayed. It can be a bit confusing. I was expecting to be prompted to plug the key after taping the buttons, not before.

Is there a way to prevent the Android pop-up "Do you allow the app to access Solo 2.3.0?" each time we need to authenticate through USB?

dschuermann, 2019-07-04:
@Mincka thanks for your testing, very interesting. I tested the SoloKey Tap successfully on Pixel 2 XL over NFC. I will do a more detailed testing with our different devices next week.

Mincka, 2019-07-04:
I notified the Solo team on Twitter on June 18th and still no official message about this specific issue. I bought 5 Solo Tap keys for my teammates and I. Nobody is able to use them with Chrome / Google and NFC, on different phone models. This is the most basic scenario that you expect to work smoothly. The comments on this thread and Twitter show that it's not an isolated issue. 

We already waited a lot more than it was usually planned. I was expecting a final product properly tested on a few major phone models by the time. In the end, I would have prefer to wait even more than being sent a non-working product. 

I am quite disappointed by the feedback at this stage: "we think that we can improve by flashing firmware" / "we may review the hardware". When? At which cost? Can we expect new hardware revision sent for free if it's not possible to make them work in standard scenarios, even after an hypothetical firmware update?

merlokk, 2019-07-04:
@Mincka test #217 please if you have hacker one

Mincka, 2019-07-04:
Don't have one but I'll be happy to test if you can provide a signed release.

merlokk, 2019-07-04:
unfortunately i cant. but it will be available in several days

ioanrogers, 2019-07-05:
Another problem phone: ZTE Axon 7
Cannot read Tap with NFC Tools normally, but can when using USB power.

When trying to register with the FIDO Example app and USB power, the Tap's led turns briefly orange but nothing happens on the phone.

conorpp, 2019-07-06:
@Mincka yes many phones were tested during prototyping, but only one particular type of Android phone was used for QA for manufacturing.  Plus we discovered a whopping 25%-30% yield issue with the AS3956, the chip we use for NFC.  Unfortunately pretty difficult to foresee issues like this.  

There are a couple issues we can fix with firmware, like when trying to register when USB powered @ioanrogers.  That way at least authenticating passively can work more reliably.

The best we/@solokeys can do at this point is make a new HW revision, with a different microcontroller than was planned for the KS campaign.  We can send free replacements to those who can't get Tap to work.  I've just started work on this and hope to start a manufacturing run by the end of this year.  Will try to get official announcement out soon.

Zjemm, 2019-07-06:
@conorpp

That would be great, let's wait for that announcement

likarum, 2019-07-11:
@conorpp
Great news!

guillaumedc, 2019-07-17:
Can't have my keys working over NFC.
It's a good news if you can find a replacement soon :-) 
Waiting for the anoucement too.

gillesw, 2019-07-22:
Trying again today on OnePlus 6t with FIDO Example : it works... Both register and authenticate.

dschuermann, 2019-07-22:
> Trying again today on OnePlus 6t with FIDO Example : it works... Both register and authenticate.

Great to hear. Does it also work for you with Google Authenticator (when you open Chrome with https://u2f.hwsecurity.dev) ?

gillesw, 2019-07-22:
> 
> 
> > Trying again today on OnePlus 6t with FIDO Example : it works... Both register and authenticate.
> 
> Great to hear. Does it also work for you with Google Authenticator (when you open Chrome with https://u2f.hwsecurity.dev) ?

On OnePlus 6T + Chrome, it works with NFC.

Did someone achieve to have NFC work on Surface GO ?

conorpp, 2019-07-22:
@gillesw awesome, what did you change?

gillesw, 2019-07-22:
Nothing... Except software update on the phone.

That's strange.

I was making tests between surface go and phone with NFC, I accidentally
put the phone on the key and bazinga...

conorpp, 2019-07-22:
Can you report which version you're on now, and which you were on previously?

gillesw, 2019-07-22:
Previously, don't know.

Currently :
Build : ONEPLUS A6013_41_190625
Oxygen OS 9.0.15
Kernel 4.9.106-perf+

Chrome 75.0.3770.143
FIDO Example 1.7

Tumeez, 2019-07-25:
USB-C and USB-A NFC doesn't work with Oneplus 3, Not even FIDO Example. When I put my Solokey to USB-C and try FIDO Example, it work, but not NFC.

Also when I open Chrome with https://u2f.hwsecurity.dev doesn't work. Not even USB-powered.

RR-O-v6.2.1-20190201-oneplus3-Unofficial-LuisROM
Kernel 3.18.124-FrancoKernel
Chrome: Newest
FIDO Example 1.7

Mincka, 2019-07-27:
One month for this issue and still no official announcement. Very disappointed by the lack of support. 😞 I am thinking about asking a refund and just buy a YubiKey 5 NFC.

conorpp, 2019-07-27:
@Mincka sorry for delay, expect official announcement this week.  In mean time, here is sign up page for customers unable to get Tap working with current phone after many attempts:
https://solokeys.us19.list-manage.com/subscribe?u=cc0c298fb99cd136bdec8294b&id=153ce13dc1

Also, can you try updating to the latest firmware?  Version 2.4.0 fixes the issues that @dschuermann found.  Hopefully it was these issues causing Samsung issues and not power.

```
# if you haven't already
pip3 install solo-python

solo key update <--secure | --hacker> --yes
```

Also with this update, you should be able to register via NFC while the key is USB-powered.

Tumeez, 2019-07-27:
My test was used newest 2.4.0 firmware.

GTB3NW, 2019-07-28:
Hey, I have a OnePlus 6 and initially had issues but worked out what was going on in the end. The antenna on the solokey is not large and the field around the phone is quite small, so you **have to place the solokey precisely into the field**. So on the oneplus 6, that's behind the camera. When the prompt in android asks you to insert your key or other OTP options, you should be able to press NFC as an option, at which point just briefly tap where the camera is and it will be picked up.

The difference between NFC on payment terminals and a solokey is that the radio on a payment terminal is much larger. Perhaps that's something that can be improved upon in future versions of solokey NFC

righettod, 2019-07-29:
Using the [FIDO Example](https://play.google.com/store/apps/details?id=de.cotech.hw.fido.example) application, its work for Register and Authenticate flow:
* SoloTap USB-C+NFC with firmware 2.4.0
* Android 9.0 on Samsung Galaxy J7

ioanrogers, 2019-07-30:
Upgraded firmware to 2.4.0, I can now register and authenticate over NFC on my Axon 7.
Thanks!

balert, 2019-08-01:
Using Oneplus 3 (Android 9) plus SoloKey USB-A NFC at firmware v2.4.2. I can register and auth via NFC while Solokey is USB powered. Without USB power neither registration nor auth work via NFC.

Mincka, 2019-08-01:
I am unable to update anymore with the web version. The current firmware is not detected ("unknown") and the update itself does not do anything. So I need to wait a USB-C / USB-A adapter to attempt to upgrade on a computer.

Today, I discover that Google is now selling their Feitan keys in France and I also received a Yubikey 5 NFC. I wanted to be sure that the issue was not related to my phone. It works perfectly every time and without even touching the back of my phone in NFC (Galaxy A8).

If the Solo update does not fix the issue, I will apply for a replacement with a revised hardware version. Hopefully it will work better.

guillaumedc, 2019-08-02:
I updated my Solo Tap (hacker) firmware top version 2.4.2 but I'm still unable to register/auth through NFC.
I doesn't work either with the USB key plugged and power supplied.

My phone is able to find the device but I get an error telling me that the communication was interrupted and I can't see the details of the NDEF tags (using NXP TagInfo).

I registered for the replacement with the revised hardware version.

Mincka, 2019-08-06:
Same here, no improvement with 2.4.2.

GTB3NW, 2019-08-07:
I'll re-iterate, the field in which you can receive NFC is not large, slowly move the device round your phone until you find the sweet spot. I believe other devices have a larger antenna, and improvement for next time perhaps. I really had to move it slowly and it was behind my camera where it worked.

mrvantage, 2019-08-07:
Updated my keys to new 2.4.2 firmware. Now I am able to use the NFC while the key is USB powered, almost every attempt works. This was not the case with the 2.3.0 firmware. I tested both the app and website from hwsecurity.dev.

However when the key is not powered, registering / authenticating only works in about 1% of the attempts., without my phone case and the key's sleeve installed. I'm feeling this might be a power related issue, like my OnePlus 3 doesn't provide enough power via NFC for the Solo Tap?

The NFC antenna in the phone is located around the camera, so identifying the sweet spot is pretty easy on my phone.

rorlork, 2019-08-15:
Same thing here, testing at u2f.bin.coffee in Samsung Galaxy S9 the Solo USB A Tap with firmware 2.4.2 can't seem to register through NFC. I get "Low level error 0x6f00" every time I try.
If I register by plugging in the usb with a usb-c adapter, then the NFC seems to work most of the time to verify.
Using the app provided by @dschuermann I can both register and verify using NFC, although to register I have to take the protecting sleeve from my phone.

datarimlens, 2019-08-16:
#223 is closed, so I try to follow up here:
My USB C Security Tap NFC arrived with 2.3 was upgraded to 2.4.2, and exhibited the following NFC behavior: on TagInfo/NXP (4.24.4):
- Taginfo goes into a loop with the Solo USB C NFC key and never finishes readout. It crashes finally or provides error slightly varying error messages depending on when the key is pulled off. It does so on both an LG G5 (on official Oreo 8.0) and a Galaxy Alpha (on official 5.0.2).
- Taginfo reads a yubikey 5 NFC on both phones without issue.
- The key reads ok with USB C connected.
Given:
- NFC works on both phones
- The Solo key reads incompletely. Yubikeys read fine.
- This does in my case not appear to be as  conorpp commented on Jul 2 an issue with the antenna coupling.
- Would the ongoing NFC issues in this thread still cover what I encounter, or would I need to classify my key as defective? Posting here for feedback, if this could still be firmware related, before opening a new issue for replacement.
Thanks.

huaracheguarache, 2019-08-19:
@dschuermann I have a Nokia 8.1 running Android 9, and when I try to register my Solo Tap over NFC in FIDO Example I get an error saying: "Internal error: UNKNOWN". This happens both when the key is powered and unpowered. If I keep holding the key against the back of the phone the app will keep trying to read the key and the error will loop. Sometimes the app crashes after many loops. Do note that I have been able to successfully use my key to sign in to Google on this phone.

I also have an old Nexus 6P running Android 8.1 that I've given a try; I am able to register and authenticate my Solo Tap both when it's powered and unpowered. However, in unpowered mode the Nexus 6P is very picky with where you place the key, and nothing happens until you find the perfect spot.

Have you guys tested the app with the Nokia 8.1?

dschuermann, 2019-08-19:
@huaracheguarache very interesting. Thanks for the report. We haven't tested it on Nokia 8.1, but on Nokia 7+. Looks like there are still enough differences. We'll look into this.

wboevink, 2019-08-31:
> So until now there were only reports regarding "OnePlus" branded smartphones.

I have the same troubles on two Galaxy S9's with two Solo Keys. No oyher problems with NFC like paying or tags etc, only with the Solo Keys

IMPULSE-255, 2019-08-31:
> 
> 
> > So until now there were only reports regarding "OnePlus" branded smartphones.
> 
> I have the same troubles on two Galaxy S9's with two Solo Keys. No oyher problems with NFC like paying or tags etc, only with the Solo Keys

also got the same issue on a Razer Phone, Sony Xperia ZX2 and occasionally on an LG G5.

[87 more replies]