solokeys/solo1 #210: Used wrong certificate in firmware, cannot update

noaccou, 2019-06-26:
I didn't have this issue, but someone could before or later.
If user will make Solo Secure from Solo Hacker by flashing Solo certificate, the key will not accept unsigned updated and updates with wrong signature.
However, in this case, user flashed wrong certificate. Private key is unknown and nobody owns it. How can user update?

**Note: Tivoization will not protect from hardware trojans/social engineering (replace the key with fake one, user could think there is an issue with service and add malicious key)/hardware disassembly and key recovery if private keys are not encrypted.**

0x0ece, 2019-06-26:
The bootloader only needs the public key to verify our signature.

My1, 2019-08-10:
regarding the first big paragraph:

if the wrong public key/certificate/whatever (to which no one had a private key) landed on the solo it would be kinda ugly, I totally agree.

a simple challenge response on lockdown with the currently known public key would help against that issue as the private key would have to be present for the lockdown.

regarding the second paragraph

replacing a user's solo with an open one to get him to add his account to it to then pull out the keys certainly is not impossible, although services could for example warn on U2F/Fido2 devices with unknown attestations and note that they may not be as secure as keys with a known attestation.

regarding hardware disassembly (this also can be seen as a reply to @0x0ece )

regarding private keys and their recovery, sure the BL only has a pub key inside but the solo obviously stores public keys on it's secure chip, and while it's definitely secure on the software side the interesting way is about physical attacks.

I mean if it's possible to get the keys out by physically tampering with the chip it would be BAD. not only would the user's device secret and resident keys be exposed but at least according to what I read about the U2F Zero the attestation is uploaded as a key-PAIR to the stick meaning if one can get the private attestation key of a secure solo the entire batch that uses that key can basically throw their attestation into oblivion (which is bad)

the only good thing about this @ghost is that to actually do this you have to essentially destroy the solo, I doubt it's possible to get it back perfectly after doing such a thing and that in its own is a big security feature as it allows me to revoke the key everywhere as soon as I notice the key missing.