solokeys/solo1 #211: Solo allow auth but only on blink DOWN

sebvallee, 2019-06-28:
Hello Solo-team,

I currently use 3 Solos (1 C, 1 A, and 1 Tap-A), and all of them have the same issue :

1. I ask for auth via Solo
2. The solo plugged in starts to flash (that seems to not be normal)
3. The auth is allowed only if I press the button, when the blink is "in low state"

At the beginning, my Solos wasn't like that, as far as I ask for auth, the LED changed color, but don't blink.

Is this a known bug ?

I tested that behavior on many computers/OSes/Solo.

Many thanks.

conorpp, 2019-07-01:
I think this is normal behavior, not a bug.  It should be about 750 ms orange and 250 ms green, and pressing-or-holding the button during the orange/not-green cycle will authorize the transaction.

sebvallee, 2019-07-02:
Should it be better (and is it possible), for the UX, to be able to press the button on all cycles (either green/orange-not-green) ? The number of times I pressed the button on the green cycle, and nothing happens makes me scream (inside, hopefully for my coworkers)…

My1, 2019-08-10:
totally agree it seems kinda weird for it literally not working 25% of the time (literally, lol)

nickray, 2019-08-11:
We should fix this.

conorpp, 2019-08-11:
I made a fix in #264.  Here is a hacker build, can one of you test it?  Pressing button gives 1 user-presence auth for a 2 second window.

[firmware-hacker-2.4.2-77-gffd854a.zip](https://github.com/solokeys/solo/files/3489862/firmware-hacker-2.4.2-77-gffd854a.zip)

```
unzip firmware-hacker-2.4.2-77-gffd854a.zip
solo program bootloader firmware-hacker-2.4.2-77-gffd854a.hex
```

My1, 2019-08-11:
I can probably in a few days when my hacker solo arrives

My1, 2019-08-13:
I have my solo hacker but testing seems a bit weird since both on stock 2.4.2 and this fw on some sites it doesnt blink, it's just yellow for example the yubico webauthn demosite.

sebvallee, 2019-09-10:
I just updated to the 2.5.2 release, and it seems that the 2.4.3 release correct this issue on my 3 solos (USBC, USBA & USBA-NFC). I close this.

closed 2019-09-10 by sebvallee