solokeys/solo1 #212: ctap_test run from ./Makefile but missing?

malcolmredheron, 2019-06-30:
I'm using master and trying to run all of the tests with `make test` but the `fido2-test` target fails because `tools/ctap_test.py` is missing. (It was deleted several months ago in https://github.com/solokeys/solo/commit/4e21c0bd8ff18c9066b88b549a54289901ae482f.)

The `test` target doesn't seem to be stale (it's used by the travis target, for example) so I figure that I must be doing something wrong but I can't figure out what. Is this file meant to be autogenerated now?

conorpp, 2019-07-01:
`tools/ctap_test.py` has been refactored into `tools/testing/main.py` tool.  It hasn't yet been added into the CI.  You can run all the tests by running the following.

```
# remove 'sim' to test actual USB device
python tools/testing/main.py sim u2f fido2 rk hid
```

malcolmredheron, 2019-07-01:
Ah, thanks. I've been trying to run that too, and getting errors so I hoped that I was running it wrong.

When I run `./main` in one terminal and `venv/bin/python tools/testing/main.py sim fido2` (or just with regular python) in another terminal, I get:

> root@3191257d5ae1:/solo# venv/bin/python tools/testing/main.py sim fido2
> Using UDP backend.
> [CtapHidDevice(localhost:8111)]
> Self test CBOR sorting
> Pass
> Self test CBOR sorting integers
> Pass
> Self test CBOR sorting major type
> Pass
> Self test CBOR sorting strings
> Pass
> Self test CBOR sorting same length strings
> Pass
> Resetting Authenticator...
> Get info
> b'\xa6\x01\x82fU2F_V2hFIDO_2_0\x02\x81khmac-secret\x03P\x88vc\x1b\xd4\xa0B\x7fWs\x0e\xc7\x1c\x9e\x02y\x04\xa4brk\xf5bup\xf5dplat\xf4iclientPin\xf4\x05\x19\x04\xb0\x06\x81\x01'
> Pass
> Traceback (most recent call last):
>   File "tools/testing/main.py", line 44, in <module>
>     FIDO2Tests(t).run()
>   File "/solo/tools/testing/tests/fido2.py", line 153, in run
>     self.test_fido2()
>   File "/solo/tools/testing/tests/fido2.py", line 1278, in test_fido2
>     self.test_get_info()
>   File "/solo/tools/testing/tests/fido2.py", line 345, in test_get_info
>     print(cbor.loads(bytes(info)))
> AttributeError: module 'fido2.cbor' has no attribute 'loads'

I have done my best to follow the build instructions but have presumably done something wrong. Is this a known mistake? I'm doing this all in a docker container (based on yours, but modified a little since I want to do more than run in-docker-build.sh) so it's easy for me to share my build steps if that's helpful.