solokeys/solo1 #213: NFC applet selection does not work correctly

dschuermann, 2019-07-01:
If we use APDU command described in the spec at https://fidoalliance.org/specs/fido-u2f-v1.2-ps-20170411/fido-u2f-nfc-protocol-v1.2-ps-20170411.html the Solokey applet selection does not work:

NfcTransport  D  NFC out: 00a4040008a0000006472f0001
NfcTransport  D  NFC  in: 6a82 ResponseApdu{data=, sw1=6a, sw2=82}

If we use "00a4040009a0000006472f000100", it works. I suspect this is due to wrong usage of memcmp in nfc.c: select_applet(). Maybe due to null terminated strings?

conorpp, 2019-07-01:
I think you're right, it looks like it's comparing an extra byte.  It's probably worked in most of my situations due to that extra memory happening to be zero.

```c
#define AID_FIDO                      "\xa0\x00\x00\x06\x47\x2f\x00\x01"
// ...
if (memcmp(aid,AID_FIDO,sizeof(AID_FIDO)) == 0)
{
     NFC_STATE.selected_applet = APP_FIDO;
     return APP_FIDO;
}
```

Will fix shortly.

merlokk, 2019-07-01:
needs to do like this
```
if length(aid) > length(aid_fido) - false
memcmp(aid, aid_fido, min(length(aid), length(aid_fido))
```
i can fix if you want)

merlokk, 2019-07-01:
https://stackoverflow.com/questions/27533193/android-hce-are-there-rules-for-aid
minimal length - 5 bytes

conorpp, 2019-07-27:
This should be resolved with latest 2.4.0 release.

closed 2019-07-27 by conorpp

dschuermann, 2019-07-27:
ACK, issue is resolved.