solokeys/solo1 #214: NFC Extended length not supported

dschuermann, 2019-07-01:
According to https://fidoalliance.org/specs/fido-u2f-v1.2-ps-20170411/fido-u2f-nfc-protocol-v1.2-ps-20170411.html#apdu-length it should be possible to use APDU extended length. This is not the case.

In the log of our FIDO client you can see that it uses a register command with extended length and gets error code 6700.

```
07-01 19:35:37.864  22929    22946           NfcTransport  D  NFC out: 00a4040009a0000006472f000100
07-01 19:35:37.872  22929    22946           NfcTransport  D  NFC  in: 5532465f56329000 ResponseApdu{data=5532465f5632, sw1=90, sw2=0}
07-01 19:35:37.873  22929    22946           NfcTransport  D  NFC communication took 8ms
07-01 19:35:37.874  22929    22946   FidoU2fAppletConne..  D  U2F applet answered correctly with version U2F_V2
07-01 19:35:37.874  22929    22946   FidoU2fAppletConne..  D  Connected to AID a0000006472f000100
07-01 19:35:37.876  22929    23079   FidoRegisterOperat..  D  challenge param: 775169a6551ecbac4c8d17561ceda65ca80cccf06894c912081c10e5192d7695
07-01 19:35:37.877  22929    23079   FidoRegisterOperat..  D  application param: abc34b4eb978b911e55240f345649cd3d7e8b583fbe066984d9881f7b5494dcb
07-01 19:35:37.877  22929    23079   FidoRegisterOperat..  D  client data: {"typ":"navigator.id.finishEnrollment","challenge":"HXUVwAc-O-BhF_Zsn3D5Hw","origin":"android:apk-key-hash:DkFg13da2wcd1HLbHNKMsr64XIQ","cid_pubkey":"unused"}
07-01 19:35:37.878  22929    23079           NfcTransport  D  NFC out: 00010300000040775169a6551ecbac4c8d17561ceda65ca80cccf06894c912081c10e5192d7695abc34b4eb978b911e55240f345649cd3d7e8b583fbe066984d9881f7b5494dcb0000
07-01 19:35:37.898  22929    23079           NfcTransport  D  NFC  in: 6700 ResponseApdu{data=, sw1=67, sw2=0}
07-01 19:35:37.899  22929    23079           NfcTransport  D  NFC communication took 20ms
07-01 19:35:38.658  22929    22929           NfcTransport  D  Nfc transport disconnected
```

Without extended length it works:
```

07-01 19:35:37.864  22929    22946           NfcTransport  D  NFC out: 00a4040009a0000006472f000100
07-01 19:35:37.872  22929    22946           NfcTransport  D  NFC  in: 5532465f56329000 ResponseApdu{data=5532465f5632, sw1=90, sw2=0}
07-01 19:35:37.873  22929    22946           NfcTransport  D  NFC communication took 8ms
07-01 19:35:37.874  22929    22946   FidoU2fAppletConne..  D  U2F applet answered correctly with version U2F_V2
07-01 19:35:37.874  22929    22946   FidoU2fAppletConne..  D  Connected to AID a0000006472f000100
07-01 19:35:37.876  22929    23079   FidoRegisterOperat..  D  challenge param: 775169a6551ecbac4c8d17561ceda65ca80cccf06894c912081c10e5192d7695
07-01 19:35:37.877  22929    23079   FidoRegisterOperat..  D  application param: abc34b4eb978b911e55240f345649cd3d7e8b583fbe066984d9881f7b5494dcb
07-01 19:35:37.877  22929    23079   FidoRegisterOperat..  D  client data: {"typ":"navigator.id.finishEnrollment","challenge":"HXUVwAc-O-BhF_Zsn3D5Hw","origin":"android:apk-key-hash:DkFg13da2wcd1HLbHNKMsr64XIQ","cid_pubkey":"unused"}
07-01 19:35:37.900  22929    23079           NfcTransport  D  NFC out: 0001030040775169a6551ecbac4c8d17561ceda65ca80cccf06894c912081c10e5192d7695abc34b4eb978b911e55240f345649cd3d7e8b583fbe066984d9881f7b5494dcb
07-01 19:35:38.511   1185     5664        WificondControl  D  Scan result ready event
07-01 19:35:38.655  22929    23079           NfcTransport  D  NFC  in: 050443df17b93f68dc9b728ebdd0bd30122d89f180212ad628b63e104be2e3ad6cb1ba203ecc709e549399c6572f44b9cb19c537db5a42f45f837806b81ea40746e5302b3ddbdea67c5b51ca1f8146394cdec558d359dd80585c4bdcd045ffd682978f9b82cb25bb2605354bfc778da5b9f431308202e93082028ea003020102020101300a06082a8648ce3d0403023081823
                                                            10b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b45523110300e060355040b0c07526f6f742043413115301306035504030c0c736f6c6f6b6579732e636f6d3121301f06092a864886f70d010901161268656c6c6f40736f6c6f6b6579732e636f6d3020170d3138313231313032323031325a180f32303
                                                            638313132383032323031325a308194310b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b455231223020060355040b0c1941757468656e74696361746f72204174746573746174696f6e3115301306035504030c0c736f6c6f6b6579732e636f6d3121301f06092a864886f70d010901161268656c6c6f4
                                                            0736f6c6f6b6579732e636f6d3059301306072a8648ce3d020106082a8648ce3d030107034200047d78f6beca40763bc75ce3acf42712c394981337a6410e92f69a3b15478db6ced9d34f3913ed127b81143be8f94c9638fee3d6cb1b5393a274f7139a0f9d5ea6a381de3081db301d0603551d0e041604149afba2210923b5e47a2a1d7a6c4e038992a30ec23081a10603551d2304819
                                                            9308196a18188a48185308182310b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b45523110300e060355040b0c07526f6f742043413115301306035504030c0c736f6c6f6b6579732e636f6d3121301f06092a864886f70d010901161268656c6c6f40736f6c6f6b6579732e636f6d820900ebd4845014a
                                                            bd15730090603551d1304023000300b0603551d0f0404030204f0300a06082a8648ce3d0403020349003046022100a17b2a1d4e42a8686d65611ef5fe6dc699ae7c208316bad6e50fd70d7e05dac90221009249f30b57d11972f2755aa2e0b6bd0f0738d0e5a24fa0f3876182d8cd48fc5730440220758578c04051330b76a39143e3ae96e646fba30b2819edf0c0a249c281f3d3d0022
                                                            06eb56dde08ba1434c37b8037d45ad9327b0238955fc13284f1af5658f469fd019000 ResponseApdu{data=050443df17b93f68dc9b728ebdd0bd30122d89f180212ad628b63e104be2e3ad6cb1ba203ecc709e549399c6572f44b9cb19c537db5a42f45f837806b81ea40746e5302b3ddbdea67c5b51ca1f8146394cdec558d359dd80585c4bdcd045ffd682978f9b82cb25bb260535
                                                            4bfc778da5b9f431308202e93082028ea003020102020101300a06082a8648ce3d040302308182310b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b45523110300e060355040b0c07526f6f742043413115301306035504030c0c736f6c6f6b6579732e636f6d3121301f06092a864886f70d0109011612
                                                            68656c6c6f40736f6c6f6b6579732e636f6d3020170d3138313231313032323031325a180f32303638313132383032323031325a308194310b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b455231223020060355040b0c1941757468656e74696361746f72204174746573746174696f6e311530130603
                                                            5504030c0c736f6c6f6b6579732e636f6d3121301f06092a864886f70d010901161268656c6c6f40736f6c6f6b6579732e636f6d3059301306072a8648ce3d020106082a8648ce3d030107034200047d78f6beca40763bc75ce3acf42712c394981337a6410e92f69a3b15478db6ced9d34f3913ed127b81143be8f94c9638fee3d6cb1b5393a274f7139a0f9d5ea6a381de3081db301d
                                                            0603551d0e041604149afba2210923b5e47a2a1d7a6c4e038992a30ec23081a10603551d23048199308196a18188a48185308182310b30090603550406130255533111300f06035504080c084d6172796c616e6431143012060355040a0c0b534f4c4f204841434b45523110300e060355040b0c07526f6f742043413115301306035504030c0c736f6c6f6b6579732e636f6d3121301f
                                                            06092a864886f70d010901161268656c6c6f40736f6c6f6b6579732e636f6d820900ebd4845014abd15730090603551d1304023000300b0603551d0f0404030204f0300a06082a8648ce3d0403020349003046022100a17b2a1d4e42a8686d65611ef5fe6dc699ae7c208316bad6e50fd70d7e05dac90221009249f30b57d11972f2755aa2e0b6bd0f0738d0e5a24fa0f3876182d8cd48
                                                            fc5730440220758578c04051330b76a39143e3ae96e646fba30b2819edf0c0a249c281f3d3d002206eb56dde08ba1434c37b8037d45ad9327b0238955fc13284f1af5658f469fd01, sw1=90, sw2=0}
07-01 19:35:38.656  22929    23079           NfcTransport  D  NFC communication took 754ms
07-01 19:35:38.658  22929    22929           NfcTransport  D  Nfc transport disconnected
```

conorpp, 2019-07-01:
I wasn't expecting extended length to be used for U2F since the requests are less than 256 bytes, but I suppose that doesn't mean extended length format won't be used.  Thanks for reporting!

@merlokk I think the issue is in `nfc_process_iblock`:

```
uint8_t * payload = buf + 1 + 5;
uint8_t plen = apdu->lc;
```

From [wikipedia](https://en.wikipedia.org/wiki/Smart_card_application_protocol_data_unit), the LC parameter can be up to 3 bytes.

> Encodes the number (Nc) of bytes of command data to follow.  0 bytes denotes Nc=01 byte with a value from 1 to 255 denotes Nc with the same value.  3 bytes, the first of which must be 0, denotes Nc in the range 1 to 65 535 (all three bytes may not be zero).

I believe we just need to handle the case when LC is 3 bytes.

merlokk, 2019-07-01:
ill check tomorrow if yubikey supports it. as i remembered iso14443 dont have extended length. let me look

merlokk, 2019-07-01:
If LE is an extended field: LC and LE must be in the same format.
https://askra.de/software/jcdocs/app-notes-2.2.2/extapdu.html

dschuermann, 2019-07-01:
Yubikey neo, yubikey 5 NFC and feitian ePass supports it.

dschuermann, 2019-07-02:
> I wasn't expecting extended length to be used for U2F since the requests are less than 256 bytes, but I suppose that doesn't mean extended length format won't be used. Thanks for reporting!

What's great about using extended length: When sending the APDU command using it, the APDU response must also be extended length. For large FIDO certificates, this saves some roundtrips, as you don't need to use ADPDU chaining by sending mutliple GET RESPONSE commands.

merlokk, 2019-07-03:
It needs to be supported.

merlokk, 2019-07-04:
@dschuermann check #217 please

conorpp, 2019-07-27:
@dschuermann Can you update to [2.4.0](https://github.com/solokeys/solo/releases/tag/2.4.0) and test again?

dschuermann, 2019-07-27:
Sry for the delay. I just got ahead and updated the solokey. It works now with extended APDU. Also the AID is now correct. Great work!

closed 2019-07-27 by dschuermann