solokeys/solo1 #216: Strange behaviour with pamu2fcfg

DanielG, 2019-07-03:
With the current 2.3.0 firmware I'm seeing quite strange behaviour on when enrolling a token with pamu2fcfg. The LED will blink orange/green instead of going solid orange waiting for a click. Correspondingly a click will only register when the led is orange.

The 2.2.2 firmware does not exhibit this behaviour, but pamu2fcfg will time out after a couple of seconds which is also less than ideal. 

----

With 2.2.2, no press:

```
USB send: 00ffffffff860008e3300ebe2712cf41000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
USB read rc read 64
USB recv: ffffffff860011e3300ebe2712cf4105000000020000000500000000000000000000000000000000000000000000000000000000000000000000000000000000
device /dev/hidraw0 discovered as 'Solo Hacker 2.2.2'
  version (Interface, Major, Minor, Build): 2, 0, 0, 0  capFlags: 5
JSON: { "challenge": "0duATrJF4CX12q3mNsqHVIoFsnp8GfPCGfqXGhE5e2M", "version": "U2F_V2", "appId": "pam:\/\/Eli" }
JSON challenge URL-B64: 0duATrJF4CX12q3mNsqHVIoFsnp8GfPCGfqXGhE5e2M
client data: { "challenge": "0duATrJF4CX12q3mNsqHVIoFsnp8GfPCGfqXGhE5e2M", "origin": "pam:\/\/Eli", "typ": "navigator.id.finishEnrollment" }
JSON: { "challenge": "0duATrJF4CX12q3mNsqHVIoFsnp8GfPCGfqXGhE5e2M", "version": "U2F_V2", "appId": "pam:\/\/Eli" }
JSON app_id pam://Eli
USB send: 000500000083004900010300000040023c613f9baa90980673e2119d53d2f3cbffbafbb1e35623e276555a6012df682a9e0ee2cedac9327cecf1d22f03b3c75a
USB write returned 65
USB send: 000500000000be53815ca358238c7c8dbe8b46450000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
now trying with timeout 1024
now trying with timeout 2048
now trying with timeout 4096
USB read rc read 64
USB rc -2
Unable to generate registration challenge, error in transport layer (-2)
```

With 2.3.0, no press:

```
USB send: 00ffffffff860008004419f17ecbcf80000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
USB read rc read 64
USB recv: ffffffff860011004419f17ecbcf8002000000020000000500000000000000000000000000000000000000000000000000000000000000000000000000000000
device /dev/hidraw0 discovered as 'Solo Hacker 2.3.0'
  version (Interface, Major, Minor, Build): 2, 0, 0, 0  capFlags: 5
JSON: { "challenge": "2S47l0tVMaSueUvH8-EgZAZ6pwl5RDrU7UuubTnPGvg", "version": "U2F_V2", "appId": "test" }
JSON challenge URL-B64: 2S47l0tVMaSueUvH8-EgZAZ6pwl5RDrU7UuubTnPGvg
client data: { "challenge": "2S47l0tVMaSueUvH8-EgZAZ6pwl5RDrU7UuubTnPGvg", "origin": "test", "typ": "navigator.id.finishEnrollment" }
JSON: { "challenge": "2S47l0tVMaSueUvH8-EgZAZ6pwl5RDrU7UuubTnPGvg", "version": "U2F_V2", "appId": "test" }
JSON app_id test
USB send: 0002000000830049000103000000407b20f76b7334028574b034997207c631a25ea8dde0e6cc1cfb1d35ef7702ee569f86d081884c7d659a2feaa0c55ad015a3
USB write returned 65
USB send: 0002000000004f1b2b0b822cd15d6c15b0f00a080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
USB read rc read 64
USB recv: 02000000830002698500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB data (len 2): 6985
USB send: 0002000000830049000103000000407b20f76b7334028574b034997207c631a25ea8dde0e6cc1cfb1d35ef7702ee569f86d081884c7d659a2feaa0c55ad015a3
USB write returned 65
USB send: 0002000000004f1b2b0b822cd15d6c15b0f00a080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
USB read rc read 64
USB recv: 02000000830002698500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB data (len 2): 6985
USB send: 0002000000830049000103000000407b20f76b7334028574b034997207c631a25ea8dde0e6cc1cfb1d35ef7702ee569f86d081884c7d659a2feaa0c55ad015a3
USB write returned 65
USB send: 0002000000004f1b2b0b822cd15d6c15b0f00a080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
USB write returned 65
now trying with timeout 2
now trying with timeout 4
now trying with timeout 8
now trying with timeout 16
now trying with timeout 32
now trying with timeout 64
now trying with timeout 128
now trying with timeout 256
now trying with timeout 512
[...]
```

conorpp, 2019-07-27:
In 2.3.0, a change in behavior for getting user consent was made.  Previously, when the token got a request, it would block and wait for a button press until replying to the platform.  But this causes premature timeout issues on some platforms (like you point out). 

The new behavior is to only wait about ~750ms and return an error that the button was not pressed, and it is up to the platform to poll the token repeatedly for the desired timeout duration.  The periods in between polls account for the blinking period.  About how long is the period for pam?

DanielG, 2019-07-28:
I think it's about a second, you can see the timeouts it's using in the output above (in ms). `512 + 256 + 128 + 64 + 32 + 16 + 8 + 4 + 2 = 1022 ms`

The thing I don't understand is how Chromium for example is managing to not have the led blink if all applications have to do this polling now.