solokeys/solo1 #218: Challenge Response in Keepass

emansih, 2019-07-05:
Hi, I'm using Keepass as my password manager. Is it possible to use my password in conjunction with button pressing on my solokey to open my DB? There is currently a [plugin](http://richardbenjaminrush.com/keechallenge/) available but it is for Yubikeys only.

youiopmop, 2019-07-16:
it's not possible at the moment, yubikey supports HMAC-SHA1 for keepass/keepassx/keepassxc, while solo only supports U2F/FIDO2

the keepassxc faq does a pretty good job of explaining why you can't use U2F/FIDO2 for it

>Q: Why only HMAC-SHA1? Why not FIDO-U2F or TOTP?
>A: Both FIDO-U2F and TOTP require a dynamic component (i.e., a counter or timestamp) for successful authentication. This is perfect for authenticating at an online service, but doesn't work for an offline database which needs to be encrypted with a fixed key. HMAC-SHA1, on the other hand, can be computed ahead of time as it only needs a fixed secret and no dynamic component of any kind. 

there was an issue about implementing yubico features (#135), but not much has been said about it. more is planned for solo though, so it *may* be supported in future (#199).

emansih, 2019-07-16:
Forgive my ignorance, is HMAC-SHA1 the same as HMAC-secret? https://github.com/solokeys/solo/pull/149

youiopmop, 2019-07-16:
unfortunately no, HMAC-secret is an extension to FIDO2, HMAC-SHA1 is different. there was talk about it in that pull request you linked
https://github.com/solokeys/solo/pull/149#issuecomment-475316832

emansih, 2019-07-22:
Thanks @youiopmop . I will close this issue.

closed 2019-07-22 by emansih