solokeys/solo1 #222: pc/device.c - created keys don't work after restarting

alphathegeek, 2019-07-13:
pc/device.c - created keys don't work after restarting software

I can successfully register a U2F key, using the udp or hidg method. When the software quits though, and I restart it , the key is recognized, but google auth says "there is a problem", and I get in a loop trying to login. the key is successful, but google will not allow a successful login.

conorpp, 2019-07-15:
Which commit are you testing at?

alphathegeek, 2019-07-15:
straight from the head/master branch, updated about 3-4 days ago.

alphathegeek, 2019-07-15:
For more information, this is using the hidg implementation, running on the usual raspberry pi development platform. I can provide debug data if necessary.

conorpp, 2019-07-15:
To confirm, this is using U2F, and not FIDO2?

conorpp, 2019-07-15:
I haven't been able to replicate from python tests, all authentications seem to work post restart of the solo program.  Can you recompile with DEBUG=1, and the following logs set in `fido2/main.c`?

```c

    set_logging_mask(
                TAG_MC |
		TAG_GEN|
		TAG_GA |
		TAG_WALLET |
		TAG_STOR |
		TAG_NFC |
		TAG_CP |
		TAG_CTAP|
		TAG_U2F|
		TAG_GREEN|
		TAG_RED|
                TAG_EXT|
		TAG_ERR
	);
```

Then post the output?

alphathegeek, 2019-07-15:
can do, will have it tomorrow.

conorpp, 2019-07-27:
@alphathegeek can you check the debug info?

alphathegeek, 2019-07-28:
Here are the Logs, this is 2 sessions:

1. Register the Key - and login, have gmail "remember" the key/computer
2. Log-back in, have google deauthorize the computer, and restart the
software. Successful login, but after the point of "remember this
computer", google says "there is a problem"..

login is not successful after that point.

// First Session -- Register the Key
[DEBUG] Using hidg backing
state file exists
[STOR] Auth state is initialized
[STOR] pin not set.
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00
[U2F] U2F_REGISTER
[U2F] sha256: c9 55 2d 6f f2 ea 7b 26 e8 b3 7d 17 4e 27 49 1f 0b 70 0d 61
a5 43 12 47 d0 f1 e7 bf fb d3 90 ad
[U2F] u2f resp: 05 04 23 5d cf 28 ba df 9e 23 4d 4e ae 19 e8 5e 21 b1 fa 36
66 4f 7e a6 e5 1b e3 7c 02 53 dc 19 c4 27 93 0f c6 12 e6 4d ef 5d 6b 52 05
98 71 42 44 fd 47 b5 9e c6 43 ac c7 e2 3b b2 f4 f8 c8 ca 65 ad 30 f8 14 39
4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23
cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 30 82 01 fb 30
82 01 a1 a0 03 02 01 02 02 01 00 30 0a 06 08 2a 86 48 ce 3d 04 03 02 30 2c
31 0b 30 09 06 03 55 04 06 13 02 55 53 31 0b 30 09 06 03 55 04 08 0c 02 4d
44 31 10 30 0e 06 03 55 04 0a 0c 07 54 45 53 54 20 43 41 30 20 17 0d 31 38
30 35 31 30 30 33 30 36 32 30 5a 18 0f 32 30 36 38 30 34 32 37 30 33 30 36
32 30 5a 30 7c 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 0b 30 09 06 03 55
04 08 0c 02 4d 44 31 0f 30 0d 06 03 55 04 07 0c 06 4c 61 75 72 65 6c 31 15
30 13 06 03 55 04 0a 0c 0c 54 45 53 54 20 43 4f 4d 50 41 4e 59 31 22 30 20
06 03 55 04 0b 0c 19 41 75 74 68 65 6e 74 69 63 61 74 6f 72 20 41 74 74 65
73 74 61 74 69 6f 6e 31 14 30 12 06 03 55 04 03 0c 0b 63 6f 6e 6f 72 70 70
2e 63 6f 6d 30 59 30 13 06 07 2a 86 48 ce 3d 02 01 06 08 2a 86 48 ce 3d 03
01 07 03 42 00 04 45 a9 02 c1 2e 9c 0a 33 fa 3e 84 50 4a b8 02 dc 4d b9 af
15 b1 b6 3a ea 8d 3f 03 03 55 65 7d 70 3f b4 02 a4 97 f4 83 b8 a6 f9 3c d0
18 ad 92 0c b7 8a 5a 3e 14 48 92 ef 08 f8 ca ea fb 32 ab 20 a3 62 30 60 30
46 06 03 55 1d 23 04 3f 30 3d a1 30 a4 2e 30 2c 31 0b 30 09 06 03 55 04 06
13 02 55 53 31 0b 30 09 06 03 55 04 08 0c 02 4d 44 31 10 30 0e 06 03 55 04
0a 0c 07 54 45 53 54 20 43 41 82 09 00 f7 c9 ec 89 f2 63 94 d9 30 09 06 03
55 1d 13 04 02 30 00 30 0b 06 03 55 1d 0f 04 04 03 02 04 f0 30 0a 06 08 2a
86 48 ce 3d 04 03 02 03 48 00 30 45 02 20 18 38 b0 45 03 69 aa a7 b7 38 62
01 af 24 97 5e 7e 74 64 1b a3 7b f7 e6 d3 af 79 28 db dc a5 88 02 21 00 cd
06 f1 e3 ab 16 21 8e d8 c0 14 af 09 4f 5b 73 ef 5e 9e 4b e7 35 eb dd 9b 6d
8f 7d f3 c4 3a d7 30 44 02 20 76 b4 46 24 83 86 f1 5f 1b 6d 1e c8 d8 50 8f
df 8c a6 f2 4a eb a0 e4 63 5b 75 c4 f2 4c 28 3a 11 02 20 3e 6b fc 49 39 48
0f d6 de b9 e0 25 7b af 1a b6 51 f2 d2 cb f3 0f 87 cb ea f7 f7 de b0 7e 61
6a 90 00
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] key handle + appid not authentic
[U2F] calc tag:
54 61 c6 d1 f9 e1 dd 70 e5 d2 58 22 80 1d 75 ee
[U2F] inp  tag:
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85
[EXT] Ignoring U2F auth request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[DEBUG] counter1: 25
[U2F] sha256: 75 db cd a7 06 55 10 60 16 c7 b5 18 76 7b 6a 78 3c b0 a9 1c
c5 97 e2 bf 01 eb dc 98 db 71 d7 7d
[U2F] u2f resp: 01 00 00 00 19 30 46 02 21 00 fa 51 23 ea 63 d4 14 03 fd 94
49 41 da 0d 05 5c 84 01 14 00 8d 3d 29 81 a5 88 c5 f1 ae 8a c8 e8 02 21 00
af 12 ef 67 d8 15 63 6f 13 76 66 0a 09 3b d0 cf 22 73 1e c5 08 92 ad 9f 30
1c a0 e7 db a5 ce cd 90 00
^CSIGINT... exiting.

// Software EXIT - and Restart Here
root@solokey:/home/pi/sk/solo# ./main
[DEBUG] Using hidg backing
state file exists
[STOR] Auth state is initialized
[STOR] pin not set.
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] key handle + appid not authentic
[U2F] calc tag:
54 61 c6 d1 f9 e1 dd 70 e5 d2 58 22 80 1d 75 ee
[U2F] inp  tag:
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85
[EXT] Ignoring U2F auth request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72
6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84
[U2F] U2F_AUTHENTICATE
[DEBUG] counter1: 25
[U2F] sha256: e2 a5 81 03 73 af 28 be f1 7f b8 77 69 c3 81 03 3d 27 87 f5
14 70 9d 9b f0 59 c8 e8 30 d5 f8 1c
[U2F] u2f resp: 01 00 00 00 19 30 45 02 20 0a 10 97 35 c4 49 ea 3d 43 d3 e3
62 de 23 6d 94 a5 5b 38 6b cb 77 45 b2 2c 17 c1 32 26 6c 31 e0 02 21 00 fc
5d 31 7e 90 ec 1f 06 a6 79 4c a5 20 73 59 ca e4 5a d5 cb cf 5c 43 18 18 53
05 98 e9 c1 6d 66 90 00

alphathegeek, 2019-07-28:
commit 2af747ddaa97e1d9dfb74c104b5298d06653addd
Merge: 9ead11d f17faca
Author: Conor Patrick <conorpp94@gmail.com>
Date:   Sat Jul 27 12:49:30 2019 -0400

    Merge pull request #229 from solokeys/fix-hmac-secret

    Fix hmac secret

alphathegeek, 2019-07-28:
Here is a successful session. It appears to work if:

1. remember computer not checked initially - causes error - does not allow login
2. check remember computer - resolves error and continues - successful login

[DEBUG] Using hidg backing
state file exists
[STOR] Auth state is initialized
[STOR] pin not set.
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] key handle + appid not authentic
[U2F] calc tag: 
54 61 c6 d1 f9 e1 dd 70 e5 d2 58 22 80 1d 75 ee 
[U2F] inp  tag: 
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[DEBUG] counter1: 25
[U2F] sha256: 8b 89 f1 d3 35 31 70 bc 49 68 fc 82 cc dd 09 08 ab 56 ff 69 53 60 fc 53 4e 7c 73 99 43 ac 9c 13 
[U2F] u2f resp: 01 00 00 00 19 30 45 02 21 00 dc d0 c8 cd 9c 03 17 d6 6f d7 da d9 43 9c d0 64 61 2a 49 4f 54 a0 42 cc 93 92 e4 94 60 06 be 14 02 20 76 84 20 cc 65 c9 02 38 f7 79 83 f0 75 bb bb 44 c6 38 eb d5 69 15 d8 18 b8 25 16 53 78 66 b1 d0 90 00 
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] key handle + appid not authentic
[U2F] calc tag: 
54 61 c6 d1 f9 e1 dd 70 e5 d2 58 22 80 1d 75 ee 
[U2F] inp  tag: 
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[DEBUG] counter1: 26
[U2F] sha256: e3 d4 3d 24 35 9f 01 f1 70 62 0c 45 01 a8 0f b0 b8 ae ca 62 f9 43 3e 21 54 34 39 da 8c 2d 15 6e 
[U2F] u2f resp: 01 00 00 00 1a 30 44 02 20 4e 39 ba d0 46 31 32 4e b9 4e ef 39 6a 51 14 50 b4 6c 1d 19 d0 d5 52 80 30 58 24 e0 20 77 d2 50 02 20 1e eb d8 c1 69 18 d2 c1 86 27 7a a8 df 3a d4 2d 03 9d 5e 0a 66 fe 98 5c 39 22 15 2d 9f f2 f0 93 90 00 
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] key handle + appid not authentic
[U2F] calc tag: 
54 61 c6 d1 f9 e1 dd 70 e5 d2 58 22 80 1d 75 ee 
[U2F] inp  tag: 
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[U2F] CHECK-ONLY
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[EXT] Ignoring U2F auth request
f8 14 39 4b 9a 13 c5 0a 0a ec 0f 9f d9 38 86 24 1a 0e 41 62 5e b7 15 6d 72 6e de 23 cf 1a b2 5d 05 61 e7 40 d7 1b 17 c8 f8 b3 be 9a f0 93 b7 84 
[U2F] U2F_AUTHENTICATE
[DEBUG] counter1: 27
[U2F] sha256: 0e df 25 e7 bb 43 66 b4 1f cd 0e 94 b9 c2 87 2f f3 f1 f7 79 8e d6 09 46 86 7f 81 0e 00 eb a9 62 
[U2F] u2f resp: 01 00 00 00 1b 30 44 02 20 14 fd 8a 7c 55 79 29 ea 94 e2 b4 39 f7 c1 08 2d aa a2 b7 f2 07 11 38 63 53 14 cc 5e ea 02 5f 08 02 20 36 35 7f 65 64 5b e1 0a 84 c3 e8 6e 02 a6 5e 61 6f be 4f ed ab 22 b2 17 36 71 69 d0 5b 5d 5a 14 90 00

alphathegeek, 2019-08-25:
related to issue #271. The global key counter in pc/device.c is not persistent across invocations. Recommend that pc/device.c store the global key counter in one of the state files, or it's own state file. Key rollback is detected by the websites, and the error messages provided by both the web-site do not provide information to indicate this.

recommend closing this.

closed 2019-08-25 by alphathegeek