solokeys/solo1 #227: I thought I had a Hacker Solo, but today I discovered it was running secure firmware. Is there any way to find out if I did this by accident?

nonnymoose, 2019-07-20:
I backed the Solo Keys kickstarter campaign and ordered a Hacker Solo. I received it near the beginning of this year, and I could have sworn that at the time, I checked and it verified as a Hacker Solo. Today, I tried to update it and it indicated that it was a Secure Solo:
```
» solo key verify
Please press the button on your Solo key
Register valid
Valid Solo Secure firmware from SoloKeys
```
What's more, it was running firmware <=1.1.0:
```
» solo key version
Firmware is out of date (key does not know the SOLO_VERSION command).
```
which means that it's pretty unlikely that I flashed the firmware at all.

I would like to determine whether this device was originally a Hacker Solo, so that I know whether this was user error or a shipping error. Is this possible

conorpp, 2019-07-27:
Yes that it a solo secure.  you can update it by running `solo key update --secure`.  For logistics issues this is more of a customer support issue so please contact hello@solokeys.com.

closed 2019-07-27 by conorpp

nonnymoose, 2019-07-27:
Is it possible to determine for certain that I did not accidentally flash an early version of the secure firmware onto a Hacker Solo?

conorpp, 2019-07-27:
Sorry i missed that.  If `solo key verify` succeeds, then it wasn't flashed incorrectly.  So it was likely shipped that way.

nonnymoose, 2019-07-27:
Thank you for clarifying! The reason I asked in a GitHub issue was that this info could be useful to someone else if they acidentally flashed the wrong firmware onto their Solo.