solokeys/solo1 #230: Trouble webupdating on Windows

gillesw, 2019-07-27:
Dear all,

Trying to web-update my Solokey A (from 2.3.0 to 2.4.0), nothing is possible : when I click on Inspect, it's asking for creating a PIN and whatever I do, nothing happens.

(Attached on Firefox, but with Chrome, it's exactly the same)
[image]

merlokk, 2019-07-27:
https://www.tenforums.com/tutorials/123007-set-up-security-key-sign-microsoft-account-microsoft-edge.html

remove security key? is the key there?

gillesw, 2019-07-27:
I'm not little-solo-keys@solokeys.com

Your link apply only for Edge.

gillesw, 2019-07-29:
I tried on another PC, same behavior. And in bootloader mode with force update, doesn't work either.

youiopmop, 2019-07-29:
some other people had issues with the webupdate on windows, the last paragraph of this comment may help

https://github.com/solokeys/solo/issues/181#issuecomment-509199785