solokeys/solo1 #234: Unexpected error code when running fido2 tests against simulator

jeamland, 2019-07-30:
Following the instructions to build/test the Solo firmware on my Mac, I ran into this:

```
(solo) benno@benno-macbook ~/w/solo> python tools/testing/main.py sim fido2
Using UDP backend.
[... lots of output ...]
Test sending zero-length pin_auth, expect PIN_NOT_SET
Send MC request with new pin auth
Pass
Send MC request with new pin auth
Pass
Pass
Traceback (most recent call last):
  File "/Users/benno/work/solo/tools/testing/tests/tester.py", line 174, in testFunc
    res = func(*args, **otherArgs)
  File "/Users/benno/.local/share/virtualenvs/solo-2GSUQ8Pd/lib/python3.7/site-packages/fido2/ctap2.py", line 670, in get_assertion
    ), timeout, AssertionResponse, on_keepalive)
  File "/Users/benno/.local/share/virtualenvs/solo-2GSUQ8Pd/lib/python3.7/site-packages/fido2/ctap2.py", line 606, in send_cbor
    raise CtapError(status)
fido2.ctap.CtapError: CTAP error: 0x35 - PIN_NOT_SET

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tools/testing/main.py", line 50, in <module>
    FIDO2Tests(t).run()
  File "/Users/benno/work/solo/tools/testing/tests/fido2.py", line 153, in run
    self.test_fido2()
  File "/Users/benno/work/solo/tools/testing/tests/fido2.py", line 1328, in test_fido2
    self.test_client_pin()
  File "/Users/benno/work/solo/tools/testing/tests/fido2.py", line 1139, in test_client_pin
    CtapError.ERR.NO_CREDENTIALS,
  File "/Users/benno/work/solo/tools/testing/tests/tester.py", line 219, in testGA
    return self.testFunc(self.ctap.get_assertion, test, *args, **kwargs)
  File "/Users/benno/work/solo/tools/testing/tests/tester.py", line 186, in testFunc
    f"Got error code {hex(e.code)}, expected {[hex(x) for x in expectedError]}"
RuntimeError: Got error code 0x35, expected ['0x33', '0x2e']
```

It appears that we're failing in the `testGA` call here around line 1132 in `tools/testing/tests/fido2.py`:

```
        with Test("Test sending zero-length pin_auth, expect PIN_NOT_SET"):
            self.testMC(
                "Send MC request with new pin auth",
                cdh,
                rp,
                user,
                key_params,
                other={"pin_auth": b"", "pin_protocol": pin_protocol},
                expectedError=CtapError.ERR.PIN_NOT_SET,
            )
            self.testGA(
                "Send MC request with new pin auth",
                rp["id"],
                cdh,
                other={"pin_auth": b"", "pin_protocol": pin_protocol},
                expectedError=[
                    CtapError.ERR.PIN_AUTH_INVALID,
                    CtapError.ERR.NO_CREDENTIALS,
                ],
            )
```

Should that `CtapError.ERR.NO_CREDENTIALS` be `CtapError.ERR.PIN_NOT_SET`?

conorpp, 2019-08-12:
Hi @jeamland,

The tests have recently been refactored and some changes have been made.  I think this particular test has been addressed.  Can you try again using our new test repo?

https://github.com/solokeys/fido2-tests

conorpp, 2019-08-15:
Closing for now since I believe this has been fixed, but please reopen if it's still a problem

closed 2019-08-15 by conorpp