solokeys/solo1 #235: STM32L432KC How to use?

Wesseldr, 2019-08-04:
Could someone please fill in the blanks please:

**Attempt 1:** Follow the manual at: https://docs.solokeys.io/solo/building/#programming
Attach a vanilla STM32L432KC board to the usb port.
1. cd targets/stm32l432
2. make build-hacker
3. solo program aux enter-bootloader
Fails at: solo.exceptions.NoSoloFoundError: no Solo found

**Attempt 2:**
1. ```cd targets/stm32l432```
2. ```make build-hacker```
3. Now open the STM32 Cube Programmer.
4. STM-LINK (right side programmer) and click connect.
5. Button Erasing & Programming,
6. Select the file: ./solo/targets/stm32l432/all.hex
7. Programm it in. (All goes well no errors.) & Run after programming
7. Led starts to blink
8. Disconnect power and plug in again.
9. run ```solo key version```
Fails at: No Solo found.

**Attempt 3:**
1. ```cd targets/stm32l432```
2. ```make build-hacker```
3. Now open the STM32 Cube Programmer.
4. STM-LINK (right side programmer) and click connect.
5. Button Erasing & Programming,
6. Select the file: ./solo/targets/stm32l432/solo.elf
7. Programm it in. (Warning Core locked...) & Run after programming
7. No blinking leds any more.
8. Disconnect power and plug in again.
9. run ```solo key version```
Fails at: No Solo found.

Expected result: The STM32L432 behaves like a solo-key-hacker on the usb plug that is on the board.

Please help, what am I missing here? Do I need an additional USB cable connected to the STM32L432KC and what pinson the board  do I have to connect to the cable ?
Or do I need to set different settings in the programmer so it starts to act like a Solo key after flashing.

And can this info please get incorporated into the Documentation. I think setting up contributors by guiding them through the steps from 0 to hero with the STM32L432KC board  is very helpful. I like to dive into the code not getting hold back by the lack of my knowledge about the development hardware used. Getting a simple step by step how to get it flashed and setup that it behaves like a Solo hacker would be very helpful.

Many thanks in advance for filling in the blanks.

JWR

Wesseldr, 2019-08-04:
One piece is solved. You **have to connect an USB cable** as follow:

* USB-D- pin to PA11(D10) 
* USB-D+ pin to PA12 (D2)
* Remove the jumper between D2 & GND after flashing
* Connect the ground and the +5v wires too.

On this USB cable to solo key will appear:
* ```solo key info``` works!
* ``` solo key version``` (does not work, don't know what wires are the button...)

Questions remain:
- Do we have to remove hat jumper of D2 & GND? because now my ST-link can no longer connect to the device. And only the attached USB works.
- What pins do we need to shortcut to do the "button pressed" signal ?

0x0ece, 2019-08-04:
FYI, a while ago we received this PR https://github.com/solokeys/solo/pull/65

I don't have a board to validate, but when working we can likely merge into our docs.

Wesseldr, 2019-08-04:
**Yes!** I can confirm that this works on the STM32L432 as described in PR #65
 
I was able to build the Solo key and do a successful authentication with it on a test website on FIDO2 including the button press by shorting wire A0 with gnd for a button press.

Think merging it would be helpful. Thank you for the link!
Case can be closed :-)

0x0ece, 2019-08-04:
Awesome - @nickray given it's confirmed we should prob merge, thoughts?

nickray, 2019-08-05:
We're just waiting on @szszszsz to remove the explicit copyright injection.

Wesseldr, 2019-08-05:
oh I see, hope he does. 

For now I'm not able to get the debugger stream flowing in OsX since it lacks udev. It is connecting but nothing comes through yet. Will have a look into that

Wesseldr, 2019-08-05:
@nickray Copyright has been removed by szszszsz 👍 
There is some merge 'code quality' problem which is odd since we're only dealing with a readme. Rest I think its ready to merge :-)

Wesseldr, 2019-08-06:
How to has been merged, closing this issue.

closed 2019-08-06 by Wesseldr