solokeys/solo1 #239: Switch to CircleCI?

nickray, 2019-08-06:
Not the first time Travis is flaky, currently: https://travis-ci.com/solokeys/solo/builds/122095134

merlokk, 2019-08-06:
it free only on linux( and
`1,000 build minutes per month`

there are 2 ways on travis:
1. use cache (but....  `Large files that are quick to install but slow to download do not benefit from caching, as they take as long to download from the cache as from the original source`)
2. reload if failed in make dependence process (needs to have external server or maybe servlet)

maybe need to search another CI that free for opensource projects ....

szszszsz, 2019-08-06:
Might be avoidable by using own Docker image as well (might either avoid cache, or have all dependencies downloaded already). A bit old example for Fedora:
- https://github.com/Nitrokey/nitrokey-fido2-firmware/blob/testing/.travis.yml
- https://github.com/Nitrokey/nitrokey-fido2-firmware/blob/testing/Dockerfile-fedora

nickray, 2019-08-07:
Travis is very eager to CI all the all-contributors merges :rofl:

nickray, 2019-08-16:
Given we're quite married to GitHub already, best option is probably to use GitHub Actions, cf. https://github.com/solokeys/solo/issues/277

(Azure Pipelines would also cover all three officially supported platforms.)

stappersg, 2019-10-18:
Cross reference to #188 
The idea is finding out if #239 and #188 are _duplicates_ ...