solokeys/solo1 #244: Recognizing all contributors

nickray, 2019-08-07:
We as SoloKeys are starting to use the [All Contributors specification](https://allcontributors.org/docs/en/specification), in recognition of the many ways this is a community effort. Code is just one part of making a project like this a success!

We will try and backfill kudos for existing contributions, but feel free to [add yourself](https://allcontributors.org/docs/en/bot/usage) by summoning the @all-contributors bot! The bot understands [quite a lot of kinds of contributions](https://allcontributors.org/docs/en/emoji-key).

A great big thank you to all of you who support our mission to make the web more secure with open source means :sparkling_heart:.

nickray, 2019-08-07:
@all-contributors, please add @merlokk for code.

nickray, 2019-08-07:
@all-contributors add @agl

nickray, 2019-08-07:
@all-contributors add @agl for bug reporting and code

nickray, 2019-08-07:
@all-contributors add @aseigler for bug https://github.com/solokeys/solo/issues/228

nickray, 2019-08-07:
@all-contributors add @dschuermann for bugs https://github.com/solokeys/solo/issues/213 and https://github.com/solokeys/solo/issues/214

nickray, 2019-08-07:
@all-contributors add @ehershey for docs

nickray, 2019-08-07:
@all-contributors add @YakBizzarro for infra

nickray, 2019-08-07:
@all-contributors add @pjz for infra and code (e.g. https://github.com/solokeys/solo/pull/17)

nickray, 2019-08-07:
@all-contributors add @yparitcher for ideas and maintenance

nickray, 2019-08-07:
@all-contributors add @StoyanDimitrov for docs

nickray, 2019-08-07:
@all-contributors add @alphathegeek for ideas

nickray, 2019-08-07:
@all-contributors add @rgerganov for ideas and code (https://github.com/solokeys/solo/issues/122 and *Linux kernel bugfixes* :astonished:)

nickray, 2019-08-07:
@all-contributors add @manuel-domke for ideas (https://github.com/solokeys/solo/issues/154) and distributing code out of Germany: https://13-37.org/de/shop/

nickray, 2019-08-07:
@all-contributors add @esden for business via distributing out of Germany and the US: https://1bitsquared.de/collections/all?page=3

nickray, 2019-08-07:
@all-contributors add @m3hm00d for docs

nickray, 2019-08-07:
@all-contributors add @szszszsz also for docs and ideas

nickray, 2019-08-22:
@allcontributors, add @kimusan for business ideas

jolo1581, 2019-10-08:
@all-contributors add @jolo1581 for Code and docs