solokeys/solo1 #262: Solokeys and Facebook

jrclooney, 2019-08-08:
On August 6th late in the evening around 7pm or 8pm EST I tried to login to Facebook using my SoloKey. I have a USB-A NFC and regular version and they were set up properly in Facebook 2.5 weeks ago. I had been happily using my key for about that same time. While not a Yubikey it does the trick for a security key and has a nice audible click when pressing the button.
So anyways I tried to login and it said in Mozilla Quantum 68 that my keys were no longer registered. I was like whaaat? So I tried signing in with an alternative way and that was okay and even trying to reregister the key again. But nope still wouldn't recognize. Then I tried Opera and it would prompt me to put in my key before saying that "Sorry an error occured" and "Try again". And the same thing with Chrome. I tried clearing out browser cookies and cache and Opera and Chrome won't bring up the pop up dialog to read security keys but Firefox will. So I don't think it is a SoloKey issue. I even tried registered another brand of security key but that I am still trying to figure out cause it's a different special kind. I know U2F and Yubikey official keys work because my other non-Facebook accounts still work. So I am wondering if my key got demagneyized or deregistered or is malfunctioning. I have a second SoloKey I haven't tested yet. And I have not tried it with a Yubikey on Facebook. Also I have not updated my key yet and used it out of the package. 
May upload screenshot later. Wonder if Facebook is pulling some hijinks or tweaking their "privacy algorithm"!!
It is odd that it doesn't work in 3 browsers.

jrclooney, 2019-08-09:
I have proof of errors here and also submitted a ticket to Facebook yesterday but no one is getting back to me.

[image]
[image]
[image]
[image]
[image]
[image]

jrclooney, 2019-08-09:
I also updated to the latest firmware after on: https://update.solokeys.com/ to the 2.4.2 version but it still said my key is not registered. Wonder if Facebook is purposely disabling 2FA with security keys for reasons.

conorpp, 2019-08-09:
@jrclooney thanks for reporting.  Did you update the firmware in between 2.5 weeks ago (when you registered) and when you ran into this issue?  Upgrading our key from `1.x.x` to `2.x.x` causes a break in registrations and they services need to be re-registered.  I'm curious if that might have happened here.

Can you try re-registering the key and seeing if authenticating works?

jrclooney, 2019-08-09:
So okay here's what I did. I updated using the update link above in Mozilla. Here's my output message:
Firefox 68.0 on Windows 10 64-bit
Your browser supports WebAuthn
Solo firmware version: 2.4.2
Latest Solo firmware version: 2.4.2
Tried to login. Then had to do my backup Facebook login. Deleted the key and reauthorized the key. Still says the key is not registered. Then I tried in Chrome:
Chrome 76.0.3809.100 on Windows 10 64-bit 
Your browser supports WebAuthn 
Solo firmware version: 2.4.2 
Latest Solo firmware version: 2.4.2
Repeated the process. Still no dice. Not working. I think it's a Facebook problem..

jrclooney, 2019-08-09:
Here is my Opera version but I didn't try to reflash in Opera since if it would have worked it should have worked in Mozilla Quantum and Chrome by now I think.
Opera 62.0.3331.116 (like Chrome 75.0.3770.142) on Windows 10 64-bit 
Your browser supports WebAuthn 
Solo firmware version: 2.4.2 
Latest Solo firmware version: 2.4.2

jrclooney, 2019-08-09:
@conorpp I also did nothing like firmware updates or anything of that sort to my registration in the last 2.5 weeks. I honestly have only logged in only once a day and so don't think it would cause device malfunction this early. I think it might be a problem with Facebook.

conorpp, 2019-08-09:
So your previous firmware version is unknown?  There could be a chance it was an older version that had the counter bug.  It works ~250 times before a byte overflows and the counter counts down.

https://github.com/solokeys/solo/releases/tag/2.0.0

So using a key prior to 2.0.0 had this bug.

jrclooney, 2019-08-09:
I ordered the solo key about 1 month ago directly from your site. I think I caught a glimpse of it saying 1.x something. But probably should have screenshotted it... So are you saying my device is bricked?

conorpp, 2019-08-09:
No, the device is not bricked.  It should work fine now with 2.0.0+ firmware, it's just some sites (like Facebook) need to be re-registered.  Orders from SoloKeys should be up to date 1-month ago but there's a chance an Amazon key was out of date.  Sorry for this sneaky bug :(

jrclooney, 2019-08-09:
Well even then after the update and re-registering it still does not seem to work. So something is still wrong I believe with Facebook. I'm just trying to figure out if it's a setting in my browser or if they disabled IP addresses. I tried using a Tunnel / VPN to see if it was locking me out based on IP and that didn't seem to be the case either. So something about Facebook interacting with my SoloKey has messed up and again I'm using 3 different USB ports on my computer. Only started having this problem 3 days ago.

youiopmop, 2019-08-09:
does your solokey work on other sites that use security keys like e.g. github,gitlab,google,twitter,etc?

jrclooney, 2019-08-09:
I will check later today. Only been using it with Facebook. But the problem I think is maybe my browser because before I even put in my SoloKey in Opera and Chrome's interface on Facebook it times out with an error. Maybe someone can investigate?

jrclooney, 2019-08-09:
Solokeys are U2F right? So can use the same key for multiple sites at once? Like Github and Facebook? They just read the key?

youiopmop, 2019-08-09:
> Solokeys are U2F right? So can use the same key for multiple sites at once? Like Github and Facebook? They just read the key?

yes, solokey supports u2f and fido2, you can use it on as many sites as you want. most sites let you add a backup key (if you have two or more of them), and/or a backup 2fa method incase you lose your main key

kd1k16oiaafz, 2019-08-09:
FYI I just had weird behaviors with my Solokeys on Facebook but it is working well on other sites (_firmware 2.4.2_). 

I successfully registered it with Chrome _v76.0.3809.100_:

[image]

But when I have tried to log on again, I had the following error:

[image]

I removed the registration and tried again but with Firefox Developer Edition _v68.0b10_ this time:

[image]

Weird fact: the device name has changed and when I have tried to log on, I had this error:

[image]

My message is in french but you can translate it into "Key not registered. Try again." (cf. @jrclooney screenshots).

jrclooney, 2019-08-09:
Yes @kd1k16oiaafz do you see what I'm talking about? Weird errors. I take it not a lot of people use U2F with Facebook but I like that they have the various 2FA methods. More than any other service I know of!

jrclooney, 2019-08-09:
Wonder if Facebook has dropped SoloKey support or something...

jrclooney, 2019-08-10:
My SoloKey also works with a Github account in Chrome but seems I have to hold it about 1.5 seconds rather than a quick tap. Anyone else having issues? Been noticing other people having issues with 2FA on Facebook also.

huaracheguarache, 2019-08-10:
@jrclooney could you try using your YubiKey when logging in to Facebook? That way we can figure out if it has something to do with Solo keys or not. I have a suspicion that Facebook messed something up and that Solo keys are not blame.

I'm also having issues logging in with my solo keys, and I get a message saying that my keys are not registered.

jrclooney, 2019-08-11:
Yep, doesn't even work with a Yubikey. Something is definitely mucked up on Facebook's side. So sad.

jrclooney, 2019-08-15:
This appears to have been fixed on the part of Facebook in the last 24 hours. Thanks for all involved. Have reached out to Facebook, Solokey. Not a Solokey issue. 
Rock on Solokey.

closed 2019-08-15 by jrclooney