solokeys/solo1 #266: Idea: command API for easy custom commands

My1, 2019-08-12:
the solo commandline software already does a few fun things like obviously programming a solo or  putting it into bootloader/dfu, reset the key and whatnot.

couldn't this be leveraged to send custom commands to a hackersolo which then can directly respond.

for example one could have a simple firmware on it which just hmacs anything given to it with an internally stored key and defines a command let's just say "hmac" for it and then that person could open a command line, say `solo cmd hmac "whatever"` and get the hmac of whatever he sent along with the solo internal key

basically this would consist of 2 parts:

1) the interface of the application allowing to send custom commands to the solo
2) the interface of the solo firmware to define commands on the solo to execute them

conorpp, 2019-08-12:
Currently hmac-secret is implemented, so a 32 or 64 byte can be hmac'd using a credential secret.

The input to the hmac would have to be hash'd down to a 32 byte value.  Currently there isn't a CLI way to interface with this though, but examples of how to do it in python are here:
https://github.com/solokeys/fido2-tests/blob/master/tests/standard/fido2/extensions/test_hmac_secret.py#L67

Happy to accept pull requests.

My1, 2019-08-12:
hmac was just an example to illustrate it without having to dive deep into complicated ideas.

conorpp, 2019-08-12:
A custom application can be added to Solo and sit on the HID "bus" and reply to custom commands sent by Solo.  

Some generic options to specify in HEX what the HID command and payload should be should be pretty easy to add.

My1, 2019-08-12:
that would be an Idea that sounds pretty useful

nickray, 2019-08-12:
We also integrated `rng` with `solo-python` as an example: https://github.com/solokeys/solo-python/blob/4375a292fd85330a5a555b7308fb21a8b91545d3/solo/cli/key.py#L30-L107

Don't forget https://github.com/solokeys/solo-python/blob/4375a292fd85330a5a555b7308fb21a8b91545d3/solo/cli/key.py#L275-L277 if you play with this.

nickray, 2019-08-12:
Another example I played with is hashes (admittedly not useful): https://github.com/solokeys/solo/blob/6c6a9bc5b697d957be3310cabd1d96f05130858d/fido2/ctaphid.c#L841-L878

If we want to do this systematically, we should allocate CTAPHID vendor commands in a somewhat systematic way.

nickray, 2019-08-12:
@My1 are you interested in implementing the HMAC?

My1, 2019-08-12:
interested sure, capable, at least not currently. and while that simple hmac was just illustrative of what an application could do with custom commands, somehting slightly more sophisticated (also using HMAC, although with SHA1) is outlined here:
https://github.com/solokeys/solo/issues/208#issuecomment-520183284