solokeys/solo1 #268: Handle button with a state machine

szszszsz, 2019-08-13:
Hi!
I have noticed issues #264 and #211. For the touch button handling we are using a state machine to make the state description more systematic. It should work for your case as well. Would you be interested in merging that?

- [Link to the POC implementation](https://github.com/Nitrokey/nitrokey-fido2-firmware/blob/button_port/targets/stm32l432/src/gpio.c) 
- [Work diagram](https://github.com/Nitrokey/nitrokey-fido-u2f-firmware/blob/master/doc/touch_button.md)

conorpp, 2019-08-13:
The current code on Solo is compatible with touch sensing (used for Somu) and it detects which to use at boot.  Happy to accept any improvements.

750ms seems really long for button to be pressed, and certainly too long for a tactile button.

conorpp, 2019-08-13:
with #264, this is the current behavior for U2F (except for the consume protection).  The behavior is [all in one short function](https://github.com/solokeys/solo/blob/cache_button/targets/stm32l432/src/device.c#L502) and while it could be clean up a bit, I think it's nice and simple.  Adding the consume protection would be good though.

For FIDO2, execution is held for UP and HID status updates are sent in parallel.  I don't think returning quickly and waiting for the platform to poll works.

conorpp, 2019-08-15:
I just refactored the button press to be a bit cleaner and:
* 1 press (activate + release) permits only one request
* Works with capacitive touch and dome.
* Permits request for a two second window (alleviates platforms that do not poll quickly).

The dome and capacitive touch state are interrupt driven and are polled to make sure they are each released before a request is permitted.

```c
    // If button was pressed within last [2] seconds, succeed.
    if (__last_button_press_time && (millis() - __last_button_press_time < 2000))
    {
        goto done;
    }

    // Set LED status and wait.
    led_rgb(0xff3520);

    // Block and wait for some time.
    ret = wait_for_button_activate(up_delay);
    if (ret) return ret;
    ret = wait_for_button_release(up_delay);
    if (ret) return ret;

    printf1(TAG_RED,"%d, %d\r\n",__last_button_press_time , millis() - __last_button_press_time);

    // If button was pressed within last [2] seconds, succeed.
    if (__last_button_press_time && (millis() - __last_button_press_time < 2000))
    {
        goto done;
    }

    // fail
    return 0;

done:
    ret = wait_for_button_release(up_delay);
    __last_button_press_time = 0;
    return 1;
```

@szszszsz Do you see any potential issues?

szszszsz, 2019-08-16:
Looks good after a brief look. I have not seen the implementation in the detail yet, so this might be inaccurate (could you link commit/file here please? or is it merged already?). Depending on the `wait_for_button_release` behavior there might be an issue:
- if it blocks forever, it suspends the device's response to the platform (does not seem to be the case, since it has time as an argument);
- if it blocks only for a specific short period, the solution will not protect user from further confirmations without his knowledge (e.g. in the event of the blocked, constant button press) after the period passes - I do not see any state handling regarding this case in the attached listing. It seems consumption is not handled correctly.

Could you elaborate, how have you handled the latter one? For reference, it is handled in the state machine I have attached originally.

conorpp, 2019-08-16:
For the button, it is activated by a (debounced) rising edge interrupt.  So if the button is held down for a long time, no request will be consumed unless another one comes when the button is finally released.  So it's still only 1 consumed.

For the capacitive touch, it is level sensitive and has the problem you point out.  I believe I can add a bit more code to make it edge sensitive which should solve it.

conorpp, 2019-08-23:
Edge trigger to touch sensing added.  Let me know if you find other issues or enhancements.

closed 2019-08-23 by conorpp