solokeys/solo1 #269: Solo Hacker: solo key hacker --update failed

sbweeden, 2019-08-14:
I am able to update my solokey hacker from a build using:

```
solo program aux enter-bootloader
solo program bootloader <build>.hex
```

However attempting to use solo key update failed with:

```
 $ solo key update --hacker
Unexpected run loop exit code: 3
We are about to update with the latest Solo Hacker firmware.
Please confirm that the connected Solo key is a Solo Hacker [y/N]: y
Wrote temporary copy of firmware-hacker-2.4.2.hex to /var/folders/6w/5fwbfgcj4wgdgh6b0tw8btvr0000gn/T/tmpdqtogtxb.hex
sha256sums coincide: 93d7e8c04b2afb8a7fb5830c7015edb7611feae1aaf999fd3715a81778c251ad
erasing firmware...
problem flashing firmware!
CTAP error: 0x01 - INVALID_COMMAND
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/solo/cli/update.py", line 212, in update
    sig = solo_client.program_file(firmware_file)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/solo/client.py", line 325, in program_file
    self.write_flash(i, data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/solo/client.py", line 191, in write_flash
    self.exchange(SoloBootloader.write, addr, data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/solo/client.py", line 142, in exchange_hid
    data = self.send_data_hid(SoloBootloader.HIDCommandBoot, req)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/solo/client.py", line 137, in send_data_hid
    return self.dev.call(cmd, data, event)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/fido2/hid.py", line 95, in call
    raise CtapError(resp[0])
fido2.ctap.CtapError: CTAP error: 0x01 - INVALID_COMMAND
```

merlokk, 2019-08-14:
need to push button on solo then insert solo to usb port, wait several flashes and release button
then restart upload

sbweeden, 2019-08-14:
Thanks - closing.

closed 2019-08-14 by sbweeden