solokeys/solo1 #271: pc/device.c counter is not persistant across multiple invocations

alphathegeek, 2019-08-15:
in pc/device.c, the counter [ global ] that keeps track of authentication requests is not persistent across "device" reboots, or invocations of the main executable. 

static int counter = 25 ; 

every time the software is started, the counter will be 25, even if the key has been previously registered. This will cause a failure, due to "key rollback" which is detected by the website.

all tests will PASS, provided the software is not terminated and restarted.

Since other state related variables are saved, please consider saving this in the state files.

conorpp, 2019-08-27:
Can you edit pc/device.c to store the counter in the state file and make a pull request?  Happy to accept.

alphathegeek, 2019-09-01:
pull request submitted. added 2 functions to r/w the global counter1.

removed the static declaration of counter1 at 25