solokeys/solo1 #287: Not working with GitHub on Android

oplik0, 2019-08-22:
Recently GitHub [added support for WebAuthn](https://github.blog/2019-08-21-github-supports-webauthn-for-security-keys/) (upgrading from older u2f api), and after re-adding my solos (tap and standard key) I can't use either to log in on Android.
Everything seems to work after pressing the button with the correct popup appearing and nfc working with solo (getting the response), but after it disappears I just get `Security key authentication failed` message on GitHub ([screenshot](https://imgur.com/a/O9JC03v)).

It doesn't work for me in both Chrome and Firefox, and using USB (with adapter) instead of NFC interface also doesn't change anything (so standard solo doesn't work as well).

The keys seems to work with other websites - or at least worked fine with webauthn.io and Twitter.

I'm using Samsung Galaxy S7.

oplik0, 2019-08-28:
@Igarron 
I tested it mostly with Solo Tap USB-A (Not sure if the nfc hardware revision is out yet but this is original version and I had no problems with it until now) with 2.4.3 firmware.

But it also doesn't seem to work with my standard Solo key (also USB-A) - also on 2.4.3 firmware - with a microUSB to USB adapter. 

The key was re-added to GitHub after they added webauthn and it was added from Windows 10 1903 (only now I thought that this might be relevant - will try to check if it works on Linux and if re-adding it from there will change anything)

conorpp, 2019-08-29:
I don't have a working Android phone at the moment, but you should be able to get some detailed info from solo using this hacker build, which is compiled with the following debug logs enabled.

[solo.zip](https://github.com/solokeys/solo/files/3555955/solo.zip)

```c
    set_logging_mask(
		/*0*/
		//TAG_GEN|
		TAG_MC |
		TAG_GA |
		TAG_WALLET |
		TAG_STOR |
		//TAG_NFC_APDU |
		TAG_NFC |
		TAG_CP |
		TAG_CTAP|
		// TAG_HID|
		TAG_U2F|
		//TAG_PARSE |
		//TAG_TIME|
		// TAG_DUMP|
		TAG_GREEN|
		TAG_RED|
                TAG_EXT|
		TAG_ERR
	);
```

Program it on a hacker.

```
unzip solo.zip
solo program bootloader solo.hex
```

Plug in via USB and open the serial port using putty / picocom / etc.  Then run the Github request on Solo over NFC while it's plugged in.  If you copy and paste the logs here that might make it clear what is going on.

lgarron, 2019-08-29:
Question: when you are registering//authenticating, how long are you holding the key to the phone?

I've recently learned that the operation will fail if I only wait for my phone to vibrate once, but succeed if I let it vibrate twice.

oplik0, 2019-08-29:
Ooh, I think I found the issue.

The key didn't work on Linux either, but unlike Android I was able to register it (on Android I have to select desktop version, go to settings from url directly and even then registration failed) - but the weird thing is that it registered as a separate key without a problem - and then it worked on android just fine.

So the issue appears to be with how Windows 10 1903 handles security keys now. It appears that from the perspective of the key a website on Windows 10 is not the same website as on Linux and Android.

That's why I got the error - from the perspective of GitHub the key wasn't linked to my account at all.

@Igarron I held it until I got the github error mostly, sometimes even longer for no reason really :) 
The length wasn't the problem it seems.

I will test it more with the logs build in a few moments, but I think I should now change the title to " Not working with GitHub on Android and Linux after registration on Windows 10"

oplik0, 2019-08-29:
A small update: the key added on Linux works with Windows too...
So I was wrong - it's not that Windows does something to domain... It seems it's just that the key registration is somehow broken.

oplik0, 2019-08-29:
Ok, got some debug logs :)
What I did here is register the key on Windows 10 > login with the key on Windows 10 > attempt to log in on Android via nfc.

```
[CTAP] cbor input structure: 0 bytes
                                    [CTAP] cbor output structure: 85 bytes.  Return 0x00
                                                                                        [CTAP] cbor input structure: 110 bytes
                                                                                                                              [CTAP] CTAP_GET_ASSERTION
                                                                                                                                                       [GA] GA map has 4 elements
                                                                                                                                                                                 [GA] GA_rpId
                                                                                                                                                                                            [GA]   ID: github.com
                    [GA] GA_clientDataHash
                                          [GA]   e0 f7 74 f4 e1 1f 17 cd 33 99 47 c2 21 f2 53 40 13 ba f1 f3 fb 59 c0 6e ec c6 97 76 60 15 86 99
                                                                                                                                                 [ERR] ../../fido2/ctap_parse.c:913: Ignoring credential is incorrect length, treating as custom
                                                   [GA] CTAP_options
                                                                    [GA] up: 0
[GA] ALLOW_LIST has 1 creds
                           [GA] CRED #0 is invalid
                                                  [GA] qsort length: 1
                                                                      [ERR] ../../fido2/ctap.c:1174: Error, no authentic credential
                                                                                                                                   [CTAP] cbor output structure: 0 bytes.  Return 0x2e
                                                                                                                                                                                      [CTAP] cbor input structure: 148 bytes
                               [CTAP] CTAP_GET_ASSERTION
                                                        [GA] GA map has 4 elements
                                                                                  [GA] GA_rpId
                                                                                              [GA]   ID: github.com
                                                                                                                   [GA] GA_clientDataHash
                                                                                                                                         [GA]   e0 f7 74 f4 e1 1f 17 cd 33 99 47 c2 21 f2 53 40 13 ba f1 f3 fb 59 c0 6e ec c6 97 76 60 15 86 99
                                                   [GA] CTAP_options
                                                                    [GA] up: 0
[GA] ALLOW_LIST has 1 creds
                           [GA] CRED #332 is invalid
                                                    [GA] qsort length: 1
                                                                        [ERR] ../../fido2/ctap.c:1174: Error, no authentic credential
                                                                                                                                     [CTAP] cbor output structure: 0 bytes.  Return 0x2e
                                                                                                                                                                                        [CTAP] cbor input structure: 5 bytes
                               [CTAP] CTAP_CLIENT_PIN
                                                     [CP] CP map has 2 elements
                                                                               [CP] CP_pinProtocol
                                                                                                  [CP] CP_subCommand
                                                                                                                    [CP] CP_cmdGetRetries
                                                                                                                                         [CTAP] cbor output structure: 3 bytes.  Return 0x00
                                                                                                                                                                                            [CTAP] cbor input structure: 0 bytes
                                   [CTAP] CTAP_GET_INFO
                                                       [CTAP] cbor output structure: 85 bytes.  Return 0x00
                                                                                                           [CTAP] cbor input structure: 5 bytes
                                                                                                                                               [CTAP] CTAP_CLIENT_PIN
[... cut ...]
[NFC] prev.Iblock: 02 00 02 03 00 61 26 63 61 1a c2 17 4a 3c c2 71 ea b0 b3 a7 dc c0 a2 74 d7 5b 78 ff 0e 69 6f c9 dd a2 60 8a 94 1e 3a eb 00 24 60 38 1c 6f 25 8e 83 95 d3 02 6f 57 1f 0d 9a 76 48 8d cd 83 76 39 b1 3a ed 31 65 60 20 fd eb 81 aa 93 c2 3f 92 8f 07 d8 52 a3 dd 65 fe 5c 2c 68 ee 7f 28 08 30 8e c2 52 79 0c 9a e4 48 00
                                                                                                                                              [NFC] Buffer being updated!
[NFC] -----
[NFC] -----
[NFC] Buffer being updated!
```

nickray, 2019-08-29:
@all-contributors add @oplik0 for bugs

oplik0, 2019-08-29:
How it looks on Linux:
Some bad logs (authentication failure)
```
[EXT] U2F_VERSION
[U2F] u2f resp: 55 32 46 5f 56 32 90 00 
[EXT] Ignoring U2F check request
fd eb 81 aa 93 c2 3f 92 8f 07 d8 52 a3 dd 65 fe 5c 2c 68 ee 7f 28 08 30 8e c2 52 79 0c 9a e4 48 
[U2F] CHECK-ONLY

[U2F] key handle + appid not authentic
[U2F] calc tag: 
12 e6 57 8e 3f 22 b6 f6 b1 bb d6 fb 3d 15 b4 65 
[U2F] inp  tag: 
fd eb 81 aa 93 c2 3f 92 8f 07 d8 52 a3 dd 65 fe 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
8a 6d 25 3e a5 c3 64 44 17 ae 0a 72 02 1d d6 4c b9 bd 42 b0 d8 f1 c9 f3 db 95 cc 8c 76 7d 50 5b c5 73 3a eb 00 24 60 38 1c 6f 25 8e 83 95 d3 02 6f 57 1f 0d 9a 76 48 8d cd 83 76 39 b1 3a ed 31 65 60 4c 01 00 00 
[U2F] key handle + appid not authentic
[U2F] calc tag: 
ec f9 34 89 58 0a 00 e5 1d ce 41 d1 e8 0e 29 20 
[U2F] inp  tag: 
8a 6d 25 3e a5 c3 64 44 17 ae 0a 72 02 1d d6 4c 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
75 07 bb 8d 01 16 74 47 9c 99 62 3e fb 9f 18 c1 74 cd 88 04 3c 56 34 0a 84 9a 80 2c 83 92 77 61 c4 1a 3a eb 00 24 60 38 1c 6f 25 8e 83 95 d3 02 6f 57 1f 0d 9a 76 48 8d cd 83 76 39 b1 3a ed 31 65 60 46 01 00 00 
[U2F] key handle + appid not authentic
[U2F] calc tag: 
b1 3e 78 36 92 5e 1d 57 9d ca 38 8a 38 1e 11 02 
75 07 bb 8d 01 16 74 47 9c 99 62 3e fb 9f 18 c1 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
fd eb 81 aa 93 c2 3f 92 8f 07 d8 52 a3 dd 65 fe 5c 2c 68 ee 7f 28 08 30 8e c2 52 79 0c 9a e4 48 
[U2F] CHECK-ONLY

[U2F] key handle + appid not authentic
[U2F] calc tag: 
80 29 0c 44 38 e0 32 b7 b8 2c 3a b6 21 b1 db 89 
[U2F] inp  tag: 
fd eb 81 aa 93 c2 3f 92 8f 07 d8 52 a3 dd 65 fe 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
8a 6d 25 3e a5 c3 64 44 17 ae 0a 72 02 1d d6 4c b9 bd 42 b0 d8 f1 c9 f3 db 95 cc 8c 76 7d 50 5b c5 73 3a eb 00 24 60 38 1c 6f 25 8e 83 95 d3 02 6f 57 1f 0d 9a 76 48 8d cd 83 76 39 b1 3a ed 31 65 60 4c 01 00 00 
[U2F] key handle + appid not authentic
22 63 f1 b5 48 05 6e 37 49 1f e1 2b 8e c8 a3 f9 
8a 6d 25 3e a5 c3 64 44 17 ae 0a 72 02 1d d6 4c 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[EXT] Ignoring U2F check request
75 07 bb 8d 01 16 74 47 9c 99 62 3e fb 9f 18 c1 74 cd 88 04 3c 56 34 0a 84 9a 80 2c 83 92 77 61 c4 1a 3a eb 00 24 60 38 1c 6f 25 8e 83 95 d3 02 6f 57 1f 0d 9a 76 48 8d cd 83 76 39 b1 3a ed 31 65 60 46 01 00 00 
[U2F] key handle + appid not authentic
0b d4 35 41 df 76 2f a2 fe 9b cc f1 3a fd ca f1 
[U2F] inp  tag: 
75 07 bb 8d 01 16 74 47 9c 99 62 3e fb 9f 18 c1 
[U2F] U2F Error code 6a80
[U2F] u2f resp: 6a 80 
[U2F] U2F_REGISTER
[U2F] U2F Error code 6985
[U2F] u2f resp: 69 85 
[U2F] U2F_REGISTER
[U2F] sha256: 32 02 66 86 6c cb 36 e6 28 61 5a 51 82 8f 97 7b 2e 31 37 0e 69 45 44 06 fc d9 8a 70 24 ff 1c 0e 
[U2F] u2f resp: 05 04 af 1e 19 3c 56 c6 a9 81 c3 9c ad 1f cc d1 49 de 35 f7 43 36 34 59 26 71 b8 15 3e 4b a3 8d ed b2 29 f1 c1 c5 f0 88 7d a8 cd fe cc b1 28 e5 36 be 44 e1 7a df 5b f3 74 c0 da 2c a3 4c 6d 53 a9 7e 30 82 11 0b 1f 1e 51 54 3c 81 4b a8 c0 23 c9 2f 18 6e d8 6e 2f e4 a5 f2 26 aa d9 43 75 e9 22 d9 19 da c5 d7 52 b7 98 c7 90 fb 57 71 a5 46 ad 1b d4 30 82 02 e9 30 82 02 8e a0 03 02 01 02 02 01 01 30 0a 06 08 2a 86 48 ce 3d 04 03 02 30 81 82 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 11 30 0f 06 03 55 04 08 0c 08 4d 61 72 79 6c 61 6e 64 31 14 30 12 06 03 55 04 0a 0c 0b 53 4f 4c 4f 20 48 41 43 4b 45 52 31 10 30 0e 06 03 55 04 0b 0c 07 52 6f 6f 74 20 43 41 31 15 30 13 06 03 55 04 03 0c 0c 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 31 21 30 1f 06 09 2a 86 48 86 f7 0d 01 09 01 16 12 68 65 6c 6c 6f 40 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 30 20 17 0d 31 38 31 32 31 31 30 32 32 30 31 32 5a 18 0f 32 30 36 38 31 31 32 38 30 32 32 30 31 32 5a 30 81 94 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 11 30 0f 06 03 55 04 08 0c 4d 61 72 79 6c 61 6e 64 31 14 30 12 06 03 55 04 0a 0c 0b 53 4f 4c 4f 20 48 41 43 4b 45 52 31 22 30 20 06 03 55 04 0b 0c 19 41 75 74 68 65 6e 74 69 63 61 74 6f 72 20 41 74 74 65 73 74 61 74 69 6f 6e 31 15 30 13 06 03 55 04 03 0c 0c 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 31 21 30 1f 06 09 2a 86 48 86 f7 0d 01 09 01 16 12 68 65 6c 6c 6f 40 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 30 59 30 13 06 07 2a 86 48 ce 3d 02 01 06 08 2a 86 48 ce 3d 03 01 07 03 42 00 04 7d 78 f6 be ca 40 76 3b c7 5c e3 ac f4 27 12 c3 94 98 13 37 a6 41 0e 92 f6 9a 3b 15 47 8d b6 ce d9 d3 4f 39 13 ed 12 7b 81 14 3b e8 f9 4c 96 38 fe e3 d6 cb 1b 53 93 a2 74 f7 13 9a 0f 9d 5e a6 a3 81 de 30 81 db 30 1d 06 03 55 1d 0e 04 16 04 14 9a fb a2 21 09 23 b5 e4 7a 2a 1d 7a 6c 4e 03 89 92 a3 0e c2 30 81 a1 06 03 55 1d 23 04 81 99 30 81 96 a1 81 88 a4 81 85 30 81 82 31 0b 30 09 06 03 55 04 06 13 02 55 53 31 11 30 0f 06 03 55 04 08 0c 08 4d 61 72 79 6c 61 6e 64 31 14 30 12 06 03 55 04 0a 0c 0b 53 4f 4c 4f 20 48 41 43 4b 45 52 31 10 30 0e 06 03 55 04 0b 0c 07 52 66f 74 20 43 41 31 15 30 13 06 03 55 04 03 0c 0c 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 31 21 30 1f 06 09 2a 86 48 86 f7 0d 01 09 01 16 12 68 65 6c 6c 6f 40 73 6f 6c 6f 6b 65 79 73 2e 63 6f 6d 82 09 00 eb d4 84 50 14 ab d1 57 30 09 06 03 55 1d 13 04 02 30 00 30 0b 06 03 55 1d 0f 04 04 03 02 04 f0 30 0a 06 08 2a 86 48 ce 3d 04 03 02 03 49 00 30 46 02 21 00 a1 7b 2a 1d 4e 42 a8 68 6d 65 61 1e f5 fe 6d c6 99 ae 7c 20 83 16 ba d6 e5 0f d7 0d 7e 05 da c9 02 21 00 92 49 f3 0b 57 d1 19 72 f2 75 5a a2 e0 b6 bd 0f 07 38 d0 e5 a2 4f a0 f3 87 61 82 d8 cd 48 fc 57 30 45 02 21 00 a3 99 86 30 7b 6e 3a a1 6d 17 2c 04 [... cut ...] a1 cf 9e a9 8c 4e 90 00 
[NFC] U2F Authenticate answered 1448919 (took 134262857)

[NFC] prev.Iblock: 03 00 02 03 00 71 20 b9 f8 c4 97 5c d9 11 2c c3 ee 34 44 26 1c 25 fb d5 59 24 b5 5f 4c 1d a7 6a 31 d2 47 5e ba f2 47 27 f1 ba 86 14 ac c4 93 82 a4 a2 24 ca 24 65 78 52 8e e8 d2 9f 8f e3 80 61 a1 cf 9e a9 8c 4e 30 e0 6d ba 31 88 53 19 df 1c 99 ab b8 2d d8 5e 90 3a b2 31 aa d5 87 e6 63 20 85 58 91 fe df 83 b5 f2 99 5c 64 5b 23 f6 0a 2c 04 e0 2e 96 11 dd 69 00 
[NFC] -----

[NFC] NFC_CMD_RBLOCK
```

conorpp, 2019-08-30:
It looks like the problem is the credential registered is a FIDO2 keyID, and then on Linux or Android, U2F is used with the FIDO2 keyID.  KeyID/credentialID is defined by the authenticator to identify a particular credential.

For Solo, FIDO2 and U2F Credential ID's are not the same.

FIDO2 & U2F keyIDs should work when used for FIDO2, but only U2F keyIDs work for U2F.  I'm not sure if I should update Solo U2F to handle FIDO2 keyIDs... I don't see why not :)

lgarron, 2019-09-03:
If you're using a key to register and authenticate on GitHub right now, all the operations use WebAuthn. We don't purposely convert the key, and we shouldn't be falling back to the U2F API.

I've ordered a Solo key, so I should hopefully be able to test soon.

JackGruber, 2019-09-04:
One Edge keys are working which are newly registered, but they no longer work with Firefox. 
Already registered keys work with Firefox, but not with Edge.

elratt0r, 2019-09-11:
Similar problem here. Registered multiple Solo Secure USB-A under Win7/Chrome with github. Tried to login in Firefox 69 (Win or Linux) does not work. Tested with the Solo Hacker, Firefox uses the U2F API to talk to the Solo Key.

lgarron, 2019-09-26:
I was able to reproduce this:

- Register in Chrome:
  - Auth works in Chrome, re-registration blocked in Chrome
  - Auth fails in Firefox, re-registration succeeds in Firefox
- Register in Firefox:
  - Auth works in Firefox, re-registration blocked in Firefox
  - Auth works in Chrome, re-registration blocked in Chrome

I haven't tried to dig into the internals, butI believe this is consistent with Firefox using U2F while Chrome uses FIDO2.

oplik0, 2019-09-26:
Interesting considering the fact that Firefox got Webauthn (and FIDO 2) support first, and I believe u2f is there just because of backwards compatibility requirements :D

lgarron, 2019-10-01:
Note:  in my comment above I'm actually just on "desktop" (macOS) for both browsers using their latest versions — that's enough to exhibit the incompatibility.

> Are you sure you have `security.webauth.webauthn` and `security.webauth.webauthn_enable_usbtoken` enabled in `about:config` in Firefox? Because it seems to register FIDO2 for me with these settings on.

These are enabled!

Is there an easy way to poke into the internals to see if FIDO2 is being used?

In any case, it's looking less like RPs can do anything to fix this, so I don't think I'll keep debugging this myself. But I'm happy to run some tests or update GitHub's WebAuthn code to be compatible if someone determines what is needed.

lgarron, 2019-10-11:
I've verified with @jcjones that:

> Firefox on MacOS supports WebAuthn level 1 with security keys that implement CTAP1.

... but Firefox on macOS doesn't support CTAP2 yet.

I don't know specifically of a browser on Windows/Android/Linux that only implements CTAP1 (Firefox implements CTAP2 on Android and with Windows Hello, I believe).

If I understand correctly, there's nothing that RPs (websites) can do to be compatible with Solo keys on all platforms using WebAuthn.

elratt0r, 2019-10-13:
I can confirm that 26db2b3f6b7f72e3461e547ef3c6c598418d6e9a fixed the problem.

Have tested it back and forth so far with Windows/Linux/Android and Firefox/Chrome.
Thanks @conorpp for fixing/enhancing this!

conorpp, 2019-10-13:
Thank you for checking it!  Will release official update soon.

oplik0, 2019-10-29:
I can also confirm that #323 fixed the issue (tested on #334 ). I think I can close this issue now.

closed 2019-10-29 by oplik0