solokeys/solo1 #308: UX: change UP, UV, UP flow

nickray, 2019-09-09:
Currently, on Windows "getAssertion with RK" checks UP, then UV (pin), then UP again. Both Yubico and Feitian don't have the initial UP check, and directly perform UV.

My1, 2019-09-09:
I am not entirely sure but maybe the initial UP maybe is to select which device you want to use. also here in linux (vivaldi on Kubuntu) I have the "touch, pin, touch again" cycle when trying demo.yubico.com (the only site I have an RK on as I don't have one with practical use currently and I don't want to waste slots as I actually use my yubi5 and don't want to reset it every whenever like my solo hacker which actually doesnt get any actual use besides dev)

after a quick check with my own sandbox (where I am pretty sure the yubi doesnt have a resident key in the first place) I also have the same 3 step thing and only after that I get to hear that there ain't no fitting key inside.

so there might also be a privacy thing, which means to not divulge whether there even is an RK on the device before performing UV&UP

so basically i think it as "select, verify, confirm"

nickray, 2019-11-02:
@conorpp is this fixed by the merged references? If so feel free to close.
To recap the issue, there's less clicking necessary with the competition, which is much better for UX.