solokeys/solo1 #312: Windows 10 Hello login

jans23, 2019-09-19:
I'm using Windows 10, Version 1903, build 18362.356 and Solokey 2.5.3. In Windows account settings I can manage "security keys", meaning resetting devices and setting their PINs. This works fine. But I can't configure my Solokey to login, I simply don't find any setting for it. @aseigler Am I missing something?

Note: I don't have an online account with Hello registered.

aseigler, 2019-09-19:
The grouping of "Security Key" in with the other options to sign into the device is either misplaced, or foreshadowing something that might be coming soon.  Currently that option is only for PIN management or fingerprint management, as well as device reset like you mention.  You would think due to the way it is grouped in the UI it would be used to configure desktop login, but I don't think that is the case...at least right now.

In 1903 a [credential provider](https://docs.microsoft.com/en-us/windows/win32/secauthn/credential-providers-in-windows) [was added](https://www.microsoft.com/en-us/microsoft-365/blog/2018/04/17/windows-hello-fido2-security-keys/) that can allow desktop login using roaming authenticators, but currently that credential provider only works if the machine is Azure AD joined, and HKEY_LOCAL_MACHINE\SOFTWARE\Policies\Microsoft\FIDO\EnableFIDODeviceLogon is present and set to 1.  It won't currently work with Hybrid, traditional AD, or workgroup joined machines.  I strongly suspect this will change [in the future](https://myignite.techcommunity.microsoft.com/sessions/81735) as it technically could work with all join modes, support is just not fully plumbed in right now.

patoberli, 2019-09-27:
Just tested it in the latest insider build of the fast channel. It sadly isn't yet working, so I don't suspect it will with the final 1909 build. It does recognize the plugged in solokey, but nothing happens when I press the button. It just stays red.
I think it's because the key/firmware is lacking an encryption or has an issue with it (or Windows Hello has...), search for hmac-secret here in the tracker.
My testmachine is not AD joined or anything, I also didn't add the registry key.

Karl-WE, 2019-10-13:
Hi is there any ETA when login with Solokey with usual Windows 10 (Microsoft Account) is going to work, just as with Yubi and eWBM?

Browser logins work fine on 1903/1909 and Edge Dev.
I've bought two keys while missing the this detail, which render them still to be used very limited. 

refer:

> https://solokeys.com/pages/faq
> Can I use Solo for password-less login in Windows 10?
> 
> Yes, Solo works with Windows Hello.  Note that currently, the Windows account needs to be configured with AAD, which likely needs to be in a business setting.
>

0x0ece, 2019-11-03:
Can you clarify?

To the best of our knowledge you can use a FIDO2 key on Windows 10 logon only in enterprise settings with AAD. Or, you can use it on web on microsoft.com and related, e.g., outlook.com.

Are you saying that you have a yubikey and you can use it to login on regular/consumer Windows 10? Do you have a Solo key to test? If that's FIDO2, then that should work. If Solo doesn't work there's likely yet-another-unclear extension, and we can build it. But first thing is to understand what this is.

rmullins08, 2019-11-22:
I have an issue trying to add the SoloKey to mysignins.microsoft.com in order to use it for AzureAD Passwordless Login.

We detected that this particular key type has been blocked by your organization. Contact your administrator for more details and try registering a different type of key.
Additional details
Correlation ID:b46c827b-01d9-4405-b619-aec8473114d9
Timestamp:2019-11-22T17:01:42Z

I haven't had issues with YubiKey Security Keys/Feitian BioPass2 with this.

fadenb, 2019-11-22:
@rmullins08 I had the same issue a while back. I believe it failed as our Azure tenant had a policy configured that required device attestation. Once we disabled that policy I was able to use the solokey to log in. A colleague told me that he could use his YubiKey while the policy was still active.

This seems to match your experience.

rmullins08, 2019-11-22:
@fadenb Thanks for the quick reply.  We do have the Enforce Attestation enabled (I think that was the default). 

I can't think of any major downsides to disabling attestation.  My users aren't advanced enough to know how to even enroll a key on their own/dumb enough to try to enroll a rogue key.

nickray, 2019-11-22:
Could said administrator not add our attestation metadata (https://github.com/solokeys/solo/tree/master/metadata)? We are working towards getting into the default whitelist, but I suspect slow progress.

aseigler, 2019-11-22:
Right now there is no option for administrator to specify custom attestation certificate.  If you select enforce, you get Microsoft's unpublished list. If you select not enforced, you can enroll any compliant authenticator, and Solo definitely works.

patoberli, 2019-11-25:
@aseigler What do you mean with "Solo works fine for consumer accounts"? It doesn't work for Windows Hello Login. Based on an article I recently read in German magazine C'T, it does work with Yubikey.

aseigler, 2019-11-25:
> @aseigler What do you mean with "Solo works fine for consumer accounts"? It doesn't work for Windows Hello Login. Based on an article I recently read in German magazine C'T, it does work with Yubikey.

Consumer accounts currently do registration through https://account.microsoft.com/security.  Right now it is for accessing web applications via a browser that uses the [WebAuthn API](https://docs.microsoft.com/en-us/windows/security/identity-protection/hello-for-business/webauthnapis).  I suspect the thing you are talking about that only works for YubiKey is the [Yubico Login for Windows](https://support.yubico.com/support/solutions/articles/15000028729-yubico-login-for-windows-configuration-guide).

> Ok, it seems it has only been enabled so far for Azure AD by Microsoft:
> https://www.jasonsamuel.com/2019/09/18/how-to-enable-fido2-password-less-authentication-with-microsoft-azure-ad-for-use-with-windows-10-and-saas-web-apps/

This is a completely different animal, for corporate, not consumer use, and registration is done through https://account.microsoft.com/security.  It uses an MSFT written [credential provider](https://docs.microsoft.com/en-us/windows/desktop/secauthn/winlogon-and-credential-providers) and works with many different make/model of security keys.

> [edit]
> revised my text

Karl-WE, 2019-11-25:
Hi there sorry for the confusion. Windows Login via FIDO2 is currently not supported or even planned.

https://twitter.com/SteveSyfuhs/status/1194733812426502144

oomek, 2021-05-21:
> Hi there sorry for the confusion. Windows Login for Home users (so Microsoft Account / MSA) via FIDO2 is currently not supported or even planned.

If I knew about this tiny detail I wouldn't have bought it.

Karl-WE, 2021-05-21:
@oomek you can use it in Windows 10 but not for Windows Machine Logon. It is possible to use it for web authentication / Windows Hello and new MS Edge for example.

oomek, 2021-05-21:
Is there a list of password managers that support Solokeys for passwordless login? Dashalane seems not.

Zanthed, 2021-05-21:
@oomek https://www.yubico.com/works-with-yubikey/catalog/

Select U2F or FIDO2/Webauthn and Password Management. No need to select a key in particular.

U2F and FIDO2 are standards, so there's absolutely no reason the software listed would not work with the Solokey except for software or hardware issues.

back2root, 2021-06-03:
@nickray is there any progress in getting into the default azure ad whitelist, maybe also any eta.?
I still don't see SoloKeys listed here: https://docs.microsoft.com/en-us/security/zero-trust/isv/fido2-hardware-vendor
:(

nickray, 2021-06-03:
We'll revisit this, yes. Also need to get Solo 2 FIDO certified in the first place.

patoberli, 2021-08-25:
Just stumbled upon this today, sadly still not supported. I wanted to use my SOLOkey in Azure. New link for the certification:
https://docs.microsoft.com/en-us/azure/active-directory/authentication/concept-fido2-hardware-vendor

timcappalli, 2021-08-25:
> Just stumbled upon this today, sadly still not supported. I wanted to use my SOLOkey in Azure. New link for the certification:
> https://docs.microsoft.com/en-us/azure/active-directory/authentication/concept-fido2-hardware-vendor

Yes, the process outlined in that article is the correct way to get added to the Azure AD allowlist.

andremae, 2021-09-17:
@nickray  Would be great to get the keys on the white list and maybe update the FAQ to highlight that solo keys are not working with AAD currently.

back2root, 2022-02-10:
> We'll revisit this, yes. Also need to get Solo 2 FIDO certified in the first place.

What is the status on this @nickray?
Looks like the key is still not listed as being fido 2 certified and solo in general isn't listed at https://docs.microsoft.com/en-us/azure/active-directory/authentication/concept-fido2-hardware-vendor#current-partners

Can you tell what is the current problem with getting the certification from the fido alliance?

nickray, 2022-03-06:
Solo 1 is certified, Solo 2 is not. We might go right for FIDO 2.1 for the latter...
With Windows partnership, there's additional Microsoft certification to do (e.g. special not-so-public rules on hmac-secret behaviour).

nickray, 2022-03-23:
FIDO MDS :)

TWEagle, 2022-05-06:
How far is it with the FIDO certified for the Solo2?

oksiquatzel, 2023-05-28:
Hello,
are there any news on this?

i would love to use my solo2 key for my windows