solokeys/solo1 #314: Readout Stores keys

jolo1581, 2019-09-25:
Hi,

I already read about resetting resistent keys, but ist there a possibility so check how many keys stored on the device.

I know it might be a Security risk. I don't want to read out the keys I only want to know how many Spacey is legt.

Greetz
Jan

nickray, 2019-09-25:
I don't think it's currently possible, so it would be a vendor extension and require extending `solo-python` too. Question is indeed whether this is a good or bad idea -- after performing UV I think it should be fine.
Would need someone to send a PR on both repos.

My1, 2019-09-27:
I think after UV it would also be not a bad idea to allow pinpoint removal of keys without having to wipe everything

nickray, 2019-09-28:
Yeah. This is really something that the FIDO Alliance should spec, rather than us and other vendors implementing our own rules and methods.

My1, 2019-09-28:
Yeah sure, but at this stage i don't even care who gets it done.

My1, 2019-11-08:
by the way yubi actually has done something about that in newer revisions of their yubikey 5 series:
https://support.yubico.com/support/solutions/articles/15000027138-yubikey-5-2-3-enhancements-to-fido-2-support

jolo1581, 2020-02-21:
> Question is indeed whether this is a good or bad idea -- after performing UV I think it should be fine.

Sorry but I am not so deep inside source Code. Can you explain what UV means.

My1, 2020-02-21:
UV = User Verification, aka entering your PIN.

yubikeys allow to get metadata for your resident keys after UV ever since 5.2.3 on both Yubikeys and Blue Yubis, which is kinda important when you wanna pinpoint remove them.

the current model with a full key being full wipe or deal with it is kinda bad

jolo1581, 2020-02-22:
OK, seems to me that it isn't very complicated to implement it as custom commands. 
Just have to check if pin is matching. 

@My1 where can I setup some resident keys for doing some tests with Implementation?

My1, 2020-02-22:
passwordless.dev for example.
use the "go usernameless too" tab, to create RKs.

ideally you use a solo you dont have important credentials on, as a reset wipes ALL credentials including the master secret for normal credentials which are partly stored on the server.

jolo1581, 2020-02-22:
I will use my solo hacker for testing 😁

riedel, 2020-02-23:
Actually you can easily test with the libfido2 fido2-token and fido2-cred utils.
`fido2-token -I -c  ` returns  `FIDO_ERR_INVALID_COMMAND`

but as expected `fido2-cred` and ` fido2-assert` both work with a residential key

riedel, 2020-02-23:
> https://support.yubico.com/support/solutions/articles/15000027138-yubikey-5-2-3-enhancements-to-fido-2-support

They are implying that it is the spec, but it is a vendor extension (code 0x41).

jolo1581, 2020-02-24:
Could be a good idea to use the same vendor extention Like yubikey. 

0x41 is free in solo

My1, 2020-02-24:
okay. btw maybe try looking at the commands chrome issues to a yubi when doing RK management and stuff, sounds weird that a vendor ext would be used there.

(do note that you need either windows earlier than 1903 or something else than Windows 10 entirely for these options in chrome)

jolo1581, 2020-02-24:
> do note that you need either windows earlier than 1903 or something else than Windows 10 entirely for these options in chrome

Sorry, I am on Manjaro

My1, 2020-02-24:
why sorry?
I said OR something else than W10 entirely.

Windows 10 1903 introduced the interception of fido calls so the options in chrome arent there. but if you have basically ANYTHING ELSE than that you are fine

riedel, 2020-02-24:
From what Yubikey says it seems to move into the 2.1 Spec: https://github.com/Yubico/libfido2/issues/129 . 

BTW: Chrome uses Yubikeys vendor extension. https://github.com/chromium/chromium/blob/997afcad45e6631e7f86bca2e18afd0218ff349d/device/fido/fido_constants.h#L228 . I don't get it!

rgerganov, 2020-03-26:
FYI, the security manager in Google Chrome works with this PR: https://github.com/solokeys/solo/issues/403

riedel, 2020-03-29:
Thanks so much. You are awesome. I have tested the last release with the libfido2 tool command line tools and it works like a charm.

Strangely I have to provide/set a PIN, but I assume this is specific to the libfido2 implementation (haven't checked the sources yet).

rgerganov, 2020-03-29:
> Strangely I have to provide/set a PIN, but I assume this is specific to the libfido2 implementation (haven't checked the sources yet).

Yes, all credential management commands require a PIN