solokeys/solo1 #316: 我的solo升级后我无法登录唯一的网站google了

kongkonga, 2019-09-27:
我的solo最近升级到2.5.3.昨天我发现无法使用它登录到google网站了。我在google帐号只有这一个登录方式。没有备用码。
我每一次版本都有更新。
我应该如何设置我的solo，使它恢复正常呢？

conorpp, 2019-09-30:
Can you share what browser and operating system you are using?

kongkonga, 2019-10-03:
> 你好，可以分享您使用的瀏覽器和操作系統嗎？
chrome  79.0.3929.0 (正式版本) canary (64 位元) 
firefox  60.9.0esr
[image]

kongkonga, 2019-10-03:
> > 你好，可以分享您使用的瀏覽器和操作系統嗎？
> > 鉻79.0.3929.0（正式版本）金絲雀（64位元）
> > 的Firefox 60.9.0esr
> > [image]

windows 10  2019LTSC