solokeys/solo1 #319: Solo Secure USB-A - Windows 10 error “This security key doesn't look familiar”

nicolazilio, 2019-10-03:
I own 2 Solo Secure keys, one USB-A and one USB-C. 

I can’t get the USB-A key to work. I use a couple of computers, both running Windows 10 (latest update). I can check and update the key at update.solokeys.com without trouble. I also have no problem setting a PIN for it (via the Windows settings app) or adding it as an option for two-factor authentication on a website, such as, for example, GitHub. 

However, when I try to use the key to confirm a login (on, for example, GitHub), I get an error message from Windows saying “This security key doesn't look familiar” and I am unable to use it. I’ve tried to set a PIN for the key, reset it and remove it, to no avail. This problem does not happen with the USB-C key. Any ideas as to what the problem could be?
 
Thank you very much in advance.

elratt0r, 2019-10-03:
I ran into a similar problem. Turned out, that I got a Solo Secure that turned to be programmed with a hacker firmware before. (Attestation Key was Hacker, Firmware was Secure - was sold as Secure).

Two options to check:
- get the solo-python package and use the verify command. That should show you if the key is genuine
- flash the stick with a hacker firmware and see if it works then on github

nicolazilio, 2019-10-05:
Thank you for the quick answer.

Right, in that case, I reckon, I should maybe be entitled to a replacement (I don't want a Hacker key).

Unfortunately, I haven't been able to test your hypothesis. I installed `solo-python` but could not detect any keys, by doing `solo ls`. Therefore, I tried to replace the USB driver using Zadig. This has had the adverse effect that now none of my keys work, even after restoring the drivers to the original version. What I mean is that, when I plug in a key, although the system does 'see' it (it shows up in the list of devices), it is not recognized anymore. That is, if I try to reflash the firmware at update.solokeys.com, the browser/windows keeps asking to plug in a key although it is plugged in already.

nicolazilio, 2019-10-05:
Quick update. After a clean install of Windows 10, the OS recognizes the key again and things also work for `solo-python`. When I run `solo key verify` I get this message `Valid Solo Secure firmware from SoloKeys`. So it would seem that is in fact a Solo Secure, and not a Hacker version(?).

kongkonga, 2019-10-06:
I hope you can have a good solution. I have a similar situation with you. When I logged in to google, I reported that I didn't have this key. I only have one key and account that requires a key. I don't have a google backup code yet. How I wish you could have a solution! ! !

kongkonga, 2019-10-06:
I hope you can find a good solution. I have a similar situation with you. When I logged into google, google reported that there is no such key. I only have one key and there is only one key account. I don't have a Google backup verification code yet. How I hope you can find a solution! ! !

conorpp, 2019-10-06:
@nicolazilio `solo key verify` reporting that means it's a correctly programmed key.  On latest versions of windows, they do not allow user-level access to security keys unless it's using the microsoft webauthn API.  solo-python uses the direct USB interface, so it needs to be run in an admin/elevated shell.

How did you register the key with Github?  What browser?  How are you trying to authenticate?  My guess is due to the heterogeneous webauthn stacks, the key registers with FIDO2 and is trying to auth with U2F.  I plan to work on an update to make our U2F work with the FIDO2 keyid's this coming week and hopefully will fix this.

nicolazilio, 2019-10-08:
@conorpp I've been doing some more troubleshooting and, in the process, I've realized that I've actually never been able to use my USB-A key on Windows.

What I added to GitHub was, in fact, PIN-based 2FA. Let me explain, the confusion originated from the fact that, on Windows, one can set up a PIN for a security key. When I tried to set up my USB-A key on GitHub, a window popped up asking me to enter my PIN. At the time, I thought that this was the PIN for my security key, as it was already plugged in, NOT the PIN that I use to sign in to Windows (and yes, they are the same, as unsafe as that is).  In fact, what I was doing was setting up my Windows login PIN as a 2FA method for GitHub.

Anyhow, shame on me! That said, I still don't fully understand why the USB-C key works and the USB-A one doesn't. In any case, I'll be eagerly waiting for the update that will fix this problem :). Thanks a lot again for the hard work.

closed 2019-10-29 by conorpp