solokeys/solo1 #325: Update page should be incorporated on Solo Start page

Karl-WE, 2019-10-13:
The https://update.solokeys.com/ can only be found via google.
it is not listed in the FAQ and on the start page

It should be listed on the start page
https://static.solokeys.com/en/start/

conorpp, 2019-10-13:
Sadly the web update won't be working on future versions of Chrome or other browsers.  So right now it's "deprecated", and I'll be working on a desktop app people can use instead.

Karl-WE, 2019-10-13:
I agree with @jolo1581 @conorpp I did everything as I should. Cannot really understand why it is deprecated still works great with Edge Dev 79.x
How the heck will people that are not technically skilled update this device?

_Originally posted by @Karl-WE in https://github.com/solokeys/solo/issues/326#issuecomment-541441881_

ccinelli, 2019-10-24:
@conorpp Why will not work? It is obviously weird that the dialog flashes multiple times and the browser may see it as something to be blocked. But you should be able to sue the browser's USB support to do it.

conorpp, 2019-10-24:
The desktop application will be more intuitive than the web application.  We'll distribute from our website, I'll look into the Windows store.

Only Chrome has USB support, and it's blocked for certain devices like security security keys.

nickray, 2019-10-24:
WebUSB is only supported by Chrome-ish, Firefox actively is against it becoming a standard.
Inside WebUSB, all HID is blacklisted, with the idea being to use WebHID (WIP).

My1, 2019-11-08:
> Can you imagine to create a Windows Store app, I am not 100% sure but PWA apps might work quite OS unrelated?

can they run as admin? in w10 1903 only admin processes can access Fido2 devices directly

patoberli, 2019-11-25:
The "old" webupdate worked fine for me in the past with several versions of Edge on Windows 10.

Karl-WE, 2019-11-25:
> The "old" webupdate worked fine for me in the past with several versions of Edge on Windows 10.

I can agree with that. Furthermore the current release of Edge Dev has fixed an issue where you would be stuck in an authentication loop between FIDO2 pin and local Windows 10 Hello PIN

My1, 2019-11-25:
depends on what edge with which w10. if it was chrome-edge on <1903, it is to be expected.