solokeys/solo1 #326: Update procedure on is wrong

Karl-WE, 2019-10-13:
> Your key is a Solo Secure!
> 
> Your firmware seems up to date! For testing, you can flash it again.
> For this, unplug your key, and plug in while keeping pressed until the LED blinks yellow.

and

>  DEVICE IS NOT IN BOOTLOADER MODE
> 
> Please unplug your key, and plug in while keeping pressed until the LED blinks yellow.
> 
> Then try to flash again
> 

in fact the USB-C and USB-A variant will not blink yellow but having a blue pulsing light

Karl-WE, 2019-10-13:
related page: https://update.solokeys.com/

jolo1581, 2019-10-13:
Maybe is coloring the wrong way to signal bootloader mode.

Maybe it would be netter using a blinking code like blinking 10 Hz or something like that.

conorpp, 2019-10-13:
Did you plug key while holding button for 2s?  LED should start off and turn flashing orange.

jolo1581, 2019-10-13:
Hi, I tested it with Solo Secure USB-C.

No bootloader:
- Not clicked --> pulsing slowly Green
- Pressing --> Pulsing slowly blue

Bootloader Mode:
- Flashing fast ... seems to me green and red ... Might be orange ;-)

But in bootloader mode it is a complete different behavior of LED.

Karl-WE, 2019-10-13:
I agree with @jolo1581 @conorpp I did everything as I should. Cannot really understand why it is deprecated still works great with Edge Dev 79.x
How the heck will people that are not technically skilled update this device?

conorpp, 2019-10-13:
It doesn't work well or at all with most browsers, I guess Edge is an exception.  It's out of our control, this is due to various OS and browser restrictions now in place or coming in the future.  I'm planning to work on a desktop application that can be used, with same-or-less amount of technical skill as the website.

conorpp, 2019-10-29:
Latest firmware doesn't require inspect step, and you can just click update.

closed 2019-10-29 by conorpp