solokeys/solo1 #335: Update to 3.0.0 release not working

clyso, 2019-10-29:
tried as stated on release description:
solo program bootloader firmware-3.0.0.json - get following error:
Not using FIDO2 interface.
using signature version >2.3.0
erasing firmware...
updated firmware 100%
time: 7.27 s
Traceback (most recent call last):
  File "/usr/local/bin/solo", line 10, in <module>
    sys.exit(solo_cli())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/solo/cli/program.py", line 172, in bootloader
    raise e
  File "/usr/local/lib/python3.7/site-packages/solo/cli/program.py", line 167, in bootloader
    p.program_file(firmware)
  File "/usr/local/lib/python3.7/site-packages/solo/client.py", line 391, in program_file
    self.verify_flash(sig)
  File "/usr/local/lib/python3.7/site-packages/solo/client.py", line 210, in verify_flash
    self.exchange(SoloBootloader.done, 0, sig)
  File "/usr/local/lib/python3.7/site-packages/solo/client.py", line 150, in exchange_hid
    raise CtapError(ret)
fido2.ctap.CtapError: CTAP error: 0x27 - OPERATION_DENIED

tried as stated on solo-python readme:
solo key update --secure
Error: no such option: --secure
tried as well
solo key update
bootloader is verifying signature...
...error!

Your key did not accept the firmware's signature! Possible reasons:
  * Tried to flash "hacker" firmware on custom hacker key with verifying bootloader

Currently, your key does not work. Please run update again with correct parameters

your documentation is really confusing...
please let me know the correct procedure. thanks.

conorpp, 2019-10-29:
Hmm both 1st and 3rd things you tried should have worked.

Where did you get your key?  Is it something you programmed?  What is the bootloader version (`solo key version`)?

clyso, 2019-10-29:
got my key from 13-37.org - just a standard key - did not program anything
solo key version: 2.5.3
i flashed previous version 2.5.3 successfully as key was stuck in bootloader mode

conorpp, 2019-10-29:
Thanks.  To confirm, when the key is in bootloader mode, and you run `solo key version`, it is `2.5.3`?  Just making sure since it's different from the application version.

conorpp, 2019-10-29:
Nvm, I figured it out.  There's 2 signatures in `firmware-3.0.0.json`, and the updater chooses based on the bootloader version.  The signatures are annotated with `2.3.0`, and not `2.5.3`, so I'll fix now and then it should work.

conorpp, 2019-10-29:
Please try again :)

clyso, 2019-10-29:
where can i find the updated version? btw: Solo Bootloader = 2.4.3

conorpp, 2019-10-29:
I updated the release with the correctly annotated `firmware-3.0.0.json`, so same process should work.

clyso, 2019-10-29:
works now:
solo key version
3.0.0 locked
thanks for the quick fix. btw:
shouldn't it read "pip install solo-python --upgrade" instead of "pip install solo --upgrade" in the release notes?

closed 2019-10-29 by clyso

conorpp, 2019-10-29:
you're right, will update, thanks!