solokeys/solo1 #337: Over twice as expensive as U2F Zero

leo60228, 2019-11-01:
The U2F Zero README says that this is "just as cheap as U2F Zero." I was looking for a replacement for my U2F Zero. According to my receipt, it costed $8.99. The Solo USB-A costs $20. What gives?

0x0ece, 2019-11-03:
I can't modify the README in U2F Zero, so I'd recommend opening an issue there. cc @conorpp 

Solo costs $20 because:
- it's more secure, using a more expensive STM32 with TRNG and readout protection
- it gives you 2fa and passwordless login, by support FIDO2 -- and it's certified with related costs
- it's maintained with a corporate structure behind it, to make sure the project doesn't die

Conor has written about costs & revenue for U2F Zero: https://conorpp.com/blog/designing-and-producing-2fa-tokens-to-sell-on-amazon
It's a very nice writeup, but numbers are a bit optimistic and even in that case it doesn't make it a sustainable business.

closed 2019-11-03 by 0x0ece

leo60228, 2019-11-03:
That repo is archived, so I can't open an issue there.