solokeys/solo1 #339: Button-guard for CTAP version getter

szszszsz, 2019-11-07:
Command `CTAPHID_GETVERSION` seems to not be guarded with the button press, which makes it providing additional data for fingerprinting.
It would be nice to have a button-guard for it for increased privacy.

My1, 2019-11-08:
what kinda of version does it output, just whether it's CTAP1/2?

szszszsz, 2019-11-08:
> Websites cannot issue HID commands through FIDO API. It is potentially additional info for fingerprinting. It's main use is to silently tell if the device needs a firmware update, so requiring a button press would hamper the UX. I'm for keeping it silent.

@conorpp  Sure, I completely understand that it is required for the good update UX. 
I might have confused the commands in the source code (as marked in #331) - I meant the version getter, that was used in the webupdate ([1]) application, exposed over FIDO U2F/FIDO2. What I noticed is, that its call execution goes silently (by design), which I thought could be used as well by malicious sites to do the potential tracking, hence the proposition to add the button check.

@My1 This is the full firmware version, e.g. `2.5.3`.

[1]: https://update.solokeys.com/

My1, 2019-11-08:
I see. well when it's only the fw, which in like 90% of all cases should be whatever is recent, I dont think it's THAT much of a problem

conorpp, 2019-11-08:
I'd be okay with adding button to the version from the web-bridge, or even just removing the version from the web-bridge.

My1, 2019-11-08:
are there maybe other ways to ask for a version, maybe during an auth, so a user could be alerted that their solo needs an update?