solokeys/solo1 #346: Firmware 3.0.1 - wrong signature?

dabeck, 2019-12-02:
I've tried to update my SoloKey to firmware 3.0.1 which results in a `bad signature` error.

My1, 2020-01-16:
check your solo tool is up to date, they made some updates to signatures a while ago where there are 2 different signatures depending on the age (actually bootloader version) of the solo, because the newer one has rollback protection, the older doesnt.

m0ellemeister, 2020-01-21:
I have the same problem with my Somu Key. I got my Somu Key with the hacker firmware and I was trying to switch it to the secure version. But I didn't want to have the "none-solokey" bundle firmware on my somu key. To switch from hacker to secure firmware, either install the bundle-secure-non-solokeys.hex or at least a bootloader. I did the last.

1. At first I created a new Python virtual environment for the solo tools, activated the venv and installed the solo tools:
```
python3 -m venv venv
. ./venv/bin/activate.fish
pip install solo-python
 ```

2. I downloaded bootloader-verifying-3.0.1.hex and verified the sha256 checksum

3. Entered bootloader mode
```
solo program aux enter-bootloader
Solo rebooted.  Reconnecting...
Not using FIDO2 interface.
```
4. entered DFU mode
```
solo program aux enter-dfu
Not using FIDO2 interface.
Please powercycle the device (pull out, plug in again)
```
5. did powercycle of the SomuKey

6. installed the bootload
```
solo program dfu bootloader-verifying-3.0.1.hex 
erasing...
downloading 100.00%  08036008 - 08036013 ...         
time: 1401 ms
verifying...
reading 100.00%  08036008 - 08036013 ...         
firmware readback verified.
```

7. left DFU mode
```
solo program aux leave-dfu
Rewriting option bytes...
Warning: OSError with write_page
Please powercycle the device (pull out, plug in again)
```
Maybe I should have had better re-installed the bootloader after this OSError? I proceeded at this point.

8. Updated the firmware of the SomuKey
```
solo key update 
Not using FIDO2 interface.
Wrote temporary copy of firmware-3.0.1.json to /tmp/tmpllpphtuz.json
sha256sums coincide: 3a8d9cd4ad48da94f62453ee9e6881eb9c6363c9544d47101e3a78d64f6bc51e
using signature version >2.5.3
erasing firmware...
updated firmware 100%             
time: 7.34 s
bootloader is verifying signature...
...pass!

Congratulations, your key was updated to the latest firmware version: 3.0.1
```
Hm... bootloader did successfully verify the signature.

9. On my Linux box I can't use the SomuKey, because of signature verify error. When I start pamu2fcfg, the SomuKey flashes for indicating that it's waiting for key press.
```
pamu2fcfg
error: (-7) Unable to verify signature
```
With NextCloud this SomuKey isn't usable too.

When I plug the SomuKey in, it's identified as Solo 3.0.1
```
kernel: usb 2-4.1.1: new full-speed USB device number 45 using xhci_hcd
kernel: usb 2-4.1.1: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 1.00
kernel: usb 2-4.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 2-4.1.1: Product: Solo 3.0.1
kernel: usb 2-4.1.1: Manufacturer: SoloKeys
kernel: usb 2-4.1.1: SerialNumber: 20823979384B
kernel: hid-generic 0003:0483:A2CA.0032: hiddev5,hidraw13: USB HID v1.11 Device [SoloKeys Solo 3.0.1] on usb-0000:00:14.0-4.1.1/input0
```
```
lsusb | grep "0483"
Bus 002 Device 045: ID 0483:a2ca STMicroelectronics
```

```

I don't have this problem with my Solo Tap Key after updating to 3.0.1:
```
solo key update
Not using FIDO2 interface.
Wrote temporary copy of firmware-3.0.1.json to /tmp/tmpzk_jimzo.json
sha256sums coincide: 3a8d9cd4ad48da94f62453ee9e6881eb9c6363c9544d47101e3a78d64f6bc51e
using signature version <=2.5.3
erasing firmware...
updated firmware 100%             
time: 7.32 s
bootloader is verifying signature...
...pass!

Congratulations, your key was updated to the latest firmware version: 3.0.1
```

The difference of the to update processes is "using signature version". For the older SoloKey Tap signature Version is <=2.5.3 and for the new SomuKey it's >2.5.3

My1, 2020-01-21:
> The difference of the two update processes is "using signature version". For the older SoloKey Tap signature Version is <=2.5.3 and for the new SomuKey it's >2.5.3

not exactly. it isnt a per-model thing but rather how old your individual key is.

the 2.5.3 Bootloader essentially brought a few new things like for example rollback prevention for the secure keys and that was basically fed into the manufacturing directly so that all newly manufactured keys run on the new Bootloader. in fact I have a Solo USB-A with a new Bootloader.

also if you dont have any important keys on your somu try a reset, maybe that helps. it helped me on some other weird error. also maybe try to check with `solo key verify` that the key is identified. frankly no idea what it outputs using a hacker firmware but whatever.

also do know that a plain flash to secure isnt gonna give you points in attestation as services may blacklist the fefault hacker certs and everything as known compromised as hacker devices are fully readable allowing users to read out the data which may make some think they are insecure. I hope it wont happen and I think it being kinda unlikely, but not impossible

conorpp, 2020-01-22:
If you updated the bootloader and then programmed an application, then the attestation key is still missing which I believe is causing these issues.  Unfortunately if you programmed a secure bootloader, you cannot return to DFU to program a bundle. 

The bundled build is just the merged bootloader + application + attestation key + lock settings.  Please check the [docs on making a custom build](https://docs.solokeys.io/solo/building/#building-a-complete-solo-build-application-bootloader-certificate) if you are building your own.

I realize the programming docs had a comment about just using the bootloader which is out of date / wrong -- sorry about that.  [It's been updated](https://github.com/solokeys/solo/issues/343#issuecomment-577304926).  For customer support (key replacements), you can reach out to hello@solokeys.com.

m0ellemeister, 2020-01-22:
> If you updated the bootloader and then programmed an application, then the attestation key is still missing which I believe is causing these issues. Unfortunately if you programmed a secure bootloader, you cannot return to DFU to program a bundle.
> 
> The bundled build is just the merged bootloader + application + attestation key + lock settings. Please check the [docs on making a custom build](https://docs.solokeys.io/solo/building/#building-a-complete-solo-build-application-bootloader-certificate) if you are building your own.
> 
> I realize the programming docs had a comment about just using the bootloader which is out of date / wrong -- sorry about that. [It's been updated](https://github.com/solokeys/solo/issues/343#issuecomment-577304926). For customer support (key replacements), you can reach out to [hello@solokeys.com](mailto:hello@solokeys.com).

Thanks for you quick reply and updating the documentation @conorpp. I really appreciate that. :)
At the time I placed the order, I wasn't aware of the fact, that a converted SoloKey from hacker to secure isn't equally to a stock secure SoloKey. Well, I placed the order back in August or somethin in 2019. My intention was not only to get some secure SoloKeys (by converting them) but I also believe in your product so I wanted to support you. I still want to support SoloKeys. I'll use the SoloKeys and likely I will order some more Solo secure keys in the future.

dabeck, 2020-04-15:
I'm closing this because it worked for me. Thanks

closed 2020-04-15 by dabeck