solokeys/solo1 #347: Arch Linux Solo Key Hacker C Not Detected

wakefieldite, 2019-12-11:
Just got my USB A models programmed, but I found that the USB C models are not being detected.
I've confirmed the USB C ports work, as I use them for other hardware before and after attempting to interact with the USB C models.
-- Output of solo ls with 2 USB A models --
```
XXXXXXXXXXXX: SoloKeys Solo Hacker 3.0.1-1-g5f1d61a
XXXXXXXXXXXX: SoloKeys Solo Hacker 3.0.1-1-g5f1d61a
```

Plugging in either USB C model and running solo ls yields no change in output.
USB A models work with a USB C adapter, USB C models do not work with or without a USB A adapter.
This issue persists through rebooting. Testing was done on a USB C port on a hub as well as the USB C port on the motherboard.

-- solo key update --
```
No Solo key found!

If you are on Linux, are your udev rules up to date?
Try adding a rule line such as the following:
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="a2ca", TAG+="uaccess"
For more, see https://docs.solokeys.io/solo/udev/
```

-- solo program aux enter-bootloader --
```
Traceback (most recent call last):
File "/usr/bin/solo", line 10, in <module>
sys.exit(solo_cli())
File "/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
return self.main(*args, **kwargs)
File "/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
rv = self.invoke(ctx)
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
return callback(*args, **kwargs)
File "/usr/lib/python3.8/site-packages/solo/cli/program.py", line 219, in enter_bootloader
return _enter_bootloader(serial)
File "/usr/lib/python3.8/site-packages/solo/cli/program.py", line 200, in _enter_bootloader
p = solo.client.find(serial)
File "/usr/lib/python3.8/site-packages/solo/client.py", line 52, in find
raise solo.exceptions.NoSoloFoundError("no Solo found")
solo.exceptions.NoSoloFoundError: no Solo found
```
After manually putting the devices in bootloader mode

-- solo program aux enter-dfu --
```
Traceback (most recent call last):
File "/usr/bin/solo", line 10, in <module>
sys.exit(solo_cli())
File "/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
return self.main(*args, **kwargs)
File "/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
rv = self.invoke(ctx)
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
return callback(*args, **kwargs)
File "/usr/lib/python3.8/site-packages/solo/cli/program.py", line 247, in enter_dfu
p = solo.client.find(serial)
File "/usr/lib/python3.8/site-packages/solo/client.py", line 52, in find
raise solo.exceptions.NoSoloFoundError("no Solo found")
solo.exceptions.NoSoloFoundError: no Solo found
```

-- dmesg output --
```
[ 4180.625283] usb 1-7: new full-speed USB device number 20 using xhci_hcd
[ 4190.899095] usb 1-7: New USB device found, idVendor=0483, idProduct=a2ca, bcdDevice= 2.00
[ 4190.899101] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4190.902990] usb 1-7: can't set config #1, error -71
```

-- udevadm monitor --environment --udev | grep DEVNAME --
`DEVNAME=/dev/bus/usb/001/020
`
-- cat /dev/bus/usb/001/020 --
`@�ʢ     )�2             !""�@@`

-- "echo "hello, Solo" > /dev/bus/usb/001/020 --
`bash: /dev/bus/usb/001/017: Permission denied`

-- sudo echo "hello, Solo" > /dev/bus/usb/001/020 --
`bash: /dev/bus/usb/001/017: Permission denied`

-- lsusb --
`Bus 001 Device 020: ID 0483:a2ca STMicroelectronics`

-- System Info --
Kernel: 5.4.2-1
libu2f-host
solokeys-udev-git 20190308.c7f0d05-2 (I've also tried the 70-solokeys-access.rules solution)
solo-python  0.0.20-1

jolo1581, 2019-12-11:
Do you have an USB-C to USB-A adapter?
So you could verify your Solo USB-C without using the USC-C port of your PC

wakefieldite, 2019-12-11:
A USB C to USB A adapter was used to test both USB C models I have and a USB A to USB C adapter was used to test the port with the USB A models I have. The USB A models worked just fine, while the USB C models still have this issue.

wakefieldite, 2019-12-11:
Just tested the desktop app on Windows, which detects the USB A models but not the USB C models.

wakefieldite, 2019-12-26:
@conorpp @0x0ece These devices are from the original kickstarter, I ended up giving up getting them working on MacOS, but were there any faulty chips? I'm wondering if mine are faulty. If you want, I can mail them to you to check out.

0x0ece, 2019-12-30:
Please send us an email to hello@solokeys.com, we'll coordinate.