solokeys/solo1 #348: Solo Key to unlock keypass

imp1sh, 2019-12-12:
I just bought a solo key to replace my yubikey.
I need it to unlock my keepass vault (keepassxc).

Until now I needed to enter the first part of the password, the second part came of of the key by pushing the button. I don't know if there's a better way but when there isn't, how can I achieve the same with the solo key?

nickray, 2019-12-12:
There is no static password functionality in the current key, and no plans to develop it. Our team's efforts are currently focused on:

- finishing OpenPGP functionality for current key
- new hardware, to increase security and hopefully increase phones with NFC support
- new firmware in Rust for new hardware

Therefore, unless the community contributes such functionality, or our resources increase substantially, static passwords will not happen. After our move to Rust, the hope is that the firmware will be more maintainable and extendable, so that we can start thinking about new functionality again, such as static passwords, TOTP, PIV, etc. etc.

On the other hand, the current key does support the `hmac-secret` extension, which can be used for challenge-response protocols, this is sketched in https://github.com/solokeys/solo-python#challenge-response.
So at least theoretically, KeePassXC (and all other password managers, TOTP tools, etc.) could implement this challenge-response protocol as authentication method. This would unlock KeePassXC authentication with *any* FIDO2-compliant key with `hmac-secret` support.

Additionally, after OpenPGP is released, I assume KeePassXC could also make use of some PGP-based protocol to authenticate you.

flocke, 2019-12-19:
There is an [open issue for KeePass XC](https://github.com/keepassxreboot/keepassxc/issues/3560) to implement exactly the `hmac-secret` functionality that @nickray is talking about. Once this is implemented you would be able to unlock it with your Solo.

prusnak, 2020-01-11:
Duplicate of #4013

GalaxyAllie, 2020-06-09:
Thanks @flocke ! That's one of the things keeping me tied to YubiKeys when I'd rather all my auth was based on standards!

Currently, I use the HMAC-SHA1 challenge-response function in KeePassXC so seeing a way to use Solo keys and other keys coming is great!

aozq, 2020-06-30:
I would like to strongly suggest static password support to be at the top of the feature request list, as it is the most simple and therefore the most widely functional passkey usage. I was thrilled to find a passkey that had a physical button, as the capacitive one on Yubikey takes multiple repeated attempts to wake up on Ubuntu text fields. 

Yubikey is great, that you can swap the slot 1 and 2 for the OTP and Static Password, which for some awesome reason means that if you're in a OTP field, like in KeePass2Android, it will use that, but otherwise will send a simple static password, all with the same press - otherwise you can long-press hold for the second slot. As this password can be up to 64 bits (128 would be better) and can be easily programmed with the Yubikey Personalization Tool, you can use this a user login to Ubuntu and to as many other applications as you have Yubikeys for or feel comfortable with password reuse in combination with a 2FA passkey. Unfortunately, that means its easier to gain access with a stolen key, but it is the fastest method with still pretty strong security to login to your most frequently used services. 

SoloKey will be the perfect solution for the capacitive button lag once it can type a simple static password. And I plan to buy 4, 6, 8 or more as soon as this happens. For now Im stuck with an awesome device that only does 'hmac-secret', whatever that is, that I have to research now to understand how not to get locked out of my computer if I follow the instructions for user login. Static is much more simple and better in this case, as I can simple write it down and keep it in a safe or encrypted place. Please make static passwords a priority! 

In fact, could you point me to where the starting point for developing the necessary code in the repo for this would be? I'm assuming in between the button click and the hmac-secret, to insert simple static password? Is there a long-press (or double click) function built yet? Thanks bunches and tons

rgerganov, 2020-06-30:
> In fact, could you point me to where the starting point for developing the necessary code in the repo for this would be? I'm assuming in between the button click and the hmac-secret, to insert [...]

I have already implemented that, see https://github.com/solokeys/solo/pull/446
You can test it with Solo Hacker and your feedback is welcome!

aozq, 2020-06-30:
@rgerganov  sooo awesome! Thank you!!

imp1sh, 2020-07-13:
I've got no hacker key but only a common solo key. Is there a way to use it with such regular device?
`solo key keyboard `
gives:
`Error: no such command "keyboard"`

michaelblyons, 2020-07-14:
I suspect it will be available in a future firmware release. I _think_ the non-hacker Solo keys can upgrade firmware unless they are intentionally locked, but if someone contradicts me they're probably right.

sbrl, 2020-07-16:
Once the PR gets merged and an official release is made, Solo secures will be able to update :-)

rugk, 2021-01-27:
Same problem for SoloKey v2 and the solution the makers propose: https://github.com/keepassxreboot/keepassxc/issues/3560#issuecomment-768393106