solokeys/solo1 #349: Documentation on PIN support

DevQps, 2019-12-16:
Hello,

I saw at #20 that someone before me already asked about PIN support. He linked to an answer that stated that FIDO2 has built-in support. Now I was wondering the following questions:

1. Does SoloKey implement PIN-support?
2. Is the PIN stored and checked inside the SoloKey? Or is it stored on the host? (An attacker could potentially find the PIN in memory or on harddisk this way)
3. What happens if someone enters a faulty pin many times? I can imagine that if the PIN is only 4-6 characters long. Someone could potentially bruteforce this mechanism.
4. Maybe the answers to these questions should be documented somewhere?

I really think it is an interesting project! Thanks for all the effort you put into this!

nickray, 2019-12-16:
1. Yes.
2. On the key.
3. After M wrong PINs, you have to powercycle. After N wrong PINs, you have to fully reset the key to use it again. N > M so you can't lock up the key by accident too easily. For the constants see https://github.com/solokeys/solo/blob/5f1d61a3ba4271085bb3e7d3450ce88c90fafed7/fido2/ctap.h#L128-L129
4. Pull requests welcome!

DevQps, 2019-12-16:
Great! Thanks for your informative response! If this should be documented somewhere, what do you think is the best location to do this? I could write a PR with a small section if that is desireable!

nickray, 2019-12-16:
Wherever you see fit in https://github.com/solokeys/solo/tree/master/docs/solo.

jolo1581, 2020-03-31:
I added some documentation with the answers of @nickray. See PR #413 

Think after merge, this issue can be closed.

DevQps, 2020-04-01:
@jolo1581 Thanks a lot!

nickray, 2020-04-01:
To clarify a minor detail, the PIN is not stored on the key, just its hash (https://fidoalliance.org/specs/fido-v2.0-ps-20190130/fido-client-to-authenticator-protocol-v2.0-ps-20190130.html#settingNewPin)