solokeys/solo1 #351: How to get started?

theltalpha, 2019-12-19:
Hallo!

I am not a complete noob to computers and GNU/Linux, however I fail getting my just received Solo USB-A key to work. I am using Debian GNU/Linux Buster 10.2. The documentation is a little bit confusing to me. I am looking for a brief step-by-step instruction how to get SoloKeys working.

So far I installed libu2f-udev 1.1.9. I copied this file: https://github.com/solokeys/solo/blob/master/udev/70-solokeys-access.rules, into /etc/udev/rules.d. I also copied this file: https://raw.githubusercontent.com/Yubico/libu2f-host/master/70-u2f.rules, into /etc/udev/rules.d. I ran this command: sudo udevadm control --reload-rules && sudo udevadm trigger. However, no file /dev/solokeys is created. Restarting the computer won't help.

What am I missing?

nickray, 2019-12-19:
The `/dev/solokeys` stuff is no longer generated, apologies if it's still in the docs.

Next step if you've setup these udev rules is to go to webauthn.io and see if it works.

theltalpha, 2019-12-20:
Great, it works, thank you!

closed 2019-12-20 by theltalpha