solokeys/solo1 #352: Problem using solokey as 2FA with Coinbase

marvingreenberg, 2019-12-20:
I'm using a solokey (secure).  I started adding it more places, and am using a pair of them on Google, Github, and Microsoft Hello.  I added my solokey to Coinbase as the 2fa, and it worked fine. 
 But then I was poking around and realized my firmware was ver out of date, some 1.0.x version IIRC.   So I updated to 3.0.1 (using the python updater).  I may have NEEDED to do this for the microsoft live.com registration.  I then checked that things were working on Github and Coinbase, sure no problem.

When I got home, I updated my backup key to 3.0.1.  And went to add it to Coinbase.  And it failed.  The first key was still working as 2fa since I had to use to try to login and to add a backup key.  So then, I switched the 2fa to an Authenticator App, then tried to add my FIRST key as 2fa, and now it fails the same way.  Coinbase support says ```you will need to have a security key that supports U2F/WebAuthN standard``` which as far as I understand Solokey does.  But something between 1.0.x and 3.0.1 is confusing Coinbase.

I see issue #125 which may be related.  I didn't realize I could downgrade firmware, perhaps I can do that to register with Coinbase.  Will that affect my other keys?

[image]

[image]

marvingreenberg, 2019-12-20:
So, I tried using Firefox to register the two keys (at firmware 3.0.1) and everything worked.  And after using Firefox to register, login now works on Chrome, Brave, Firefox.

Other details I didn't mention, I'm on Macos with a Solokey v1.2 USB C.

conorpp, 2019-12-20:
Upgrading from firmware that is below 2.1.0 may cause existing registrations on the key to break (authentications may have problems).  But re-registering the key solves.

Solo does support U2F/WebAuthn.  If you're using the latest firmware, it shouldn't be using a count with the MS-bit set, which caused #125. 

My guess is maybe Coinbase does not use a unique counter for each token?  Maybe re-logging in reset the coinbase counter?

marvingreenberg, 2019-12-20:
In my case, upgrading from 1.0 to 3.0 did not break my existing registration.  But it prevented REGISTERING a new key, at Coinbase, but only with Chrome (and Brave).  Our latest comments probably crossed.  I have successfully registered both keys at Coinbase using Firefox, but new registrations using Chrome were failing.  I can now login with all three browsers.

davidpgil, 2021-04-21:
I am currently trying to use my Solokey v1 with the latest firmware 4.15 on Coinbase as my 2FA device. It throws the same error window indicated on this ticket. Will this ever be ficed or is this going to be abandoned in favor of Solokey v2?

aseigler, 2021-04-21:
I was able to reproduce this issue with a new account and new registration on Coinbase.  I don't think this is a Solo issue, I think this is likely an issue on the Coinbase side.  I was able to register a YubiKey 5 with no problem, and the packed attestation looks nearly identical to the failed Solo attempt.  If I had to guess I would say they are either whitelisting authenticators based on aaguid or maybe they do not like length of the credentialID.  Firefox didn't work either though, unlike earlier report.

szszszsz, 2021-04-21:
Just FYI, I have looked into this briefly as well - confirmed that Chrome accepted the device's response and have not found any warnings or unusual messages in the its development logs.

conorpp, 2021-04-21:
I have looked into this before, and came to the same conclusion of @aseigler.  I wonder if U2F is getting used in some situations and causes it to work, but FIDO2 (containing AAGUID, different attn statement etc) runs into issues on their server.

If you all have time please kindly complain to https://twitter.com/CoinbaseSupport, hopefully they can get a ticket opened on their side.

andrewconnell, 2021-09-02:
I had the same experience as @marvingreenberg [above](https://github.com/solokeys/solo/issues/352#issuecomment-567959458)... had to use FF to register my key (v4.1.2). Once that was done, I could login via Chrome.

I'd be happy to message Coinbase support, but I'm not exactly sure what to say... :)